seemoo-lab/nexmon_csi #222: phase difference for AoA, ToF and Doppler profile

yujianyuanhaha, 2021-06-18:
greetings,

As stated in many other work in intel 5300 and Atheros chip CSI, the phase different is pretty useful for: (1) Angle of Arrival (AoA) estimation, using the phase difference in a Uniform Linear Array, and 3 antennas of a ASUS router can do so, (2), Time-of-Flight (ToF) estimation, using multiple subcarriers of the OFMD subcarriers, (3), Doppler profile using adjacent packages especially for human activity monitor.  

In another post I have note the unstable amplitude issue during cross session -- which mean the amplitude will turn into another waveform given same environment but diff time session, and it is universal across nexmon, intel 5300 and Atheros.

However, I figure there have not been solid work done with nexmon, especially for the offset calibration, [issue] mention the flip issue [issue] mention SFO issue.

Take the Raspberry PI 5G-band 80M CSI collection is a static environment.

1. the ToF estimation/ phase diff between subcarriers
[image]

2. The AoA estimation
[image]

3. The Doppler profile
[image]

yujianyuanhaha, 2021-06-20:
another work we can refer to is "Tewes, Simon, and Aydin Sezgin. "WS-WiFi: Wired Synchronization for CSI Extraction on COTS-WiFi-Transceivers." IEEE Internet of Things Journal 8.11 (2021): 9099-9108.", it mentioned the temperature and attennas imperfection that bring the offset.

haochenku, 2021-07-26:
Hi Jet, how do you make the multiple NSS work? I tried in my side, for 2nd NSS and 3rd NSS, they are all zeros. @yujianyuanhaha

yujianyuanhaha, 2021-07-27:
@haochenku you need to setup `iperf3` between AP and a Client to trigger high throughput communication, so that the nexmon router can overhear tons of CSI.