seemoo-lab/nexmon_csi #223: cable connect transmitter router and receiver router

yujianyuanhaha, 2021-06-20:
greetings,

where can we find the cable to wired connect transmitter router and receiver router, like in the paper "https://dl.acm.org/doi/pdf/10.1145/3349623.3355477". Is it safe to do so?
[image]

The reason I plan to do this is try to figure how these offset (SFO or CFO) is like, and how they  matter the phases.

Best

Jet

RenjieZhao, 2021-07-09:
Check rp-sma when you looking for the connecter. Commercial devices usually use rp-sma instead of sma. 
Suggest to add some attenuation in between, say 30dB, if you want to directly connect TX and RX. The TX power may be too high to saturate the LNA of the RX. Keep in mind that the connections will introduce some phase offsets.

yujianyuanhaha, 2021-07-29:
i later figure out coaxial Coax Adapter SMA Female to RP-SMA Male is the cheapest
[image]

all others like SMA Female cable, splitter is easy to go