seemoo-lab/nexmon_csi #226: How can I distiguish the CSI data of the 4 antenna's on the ASUS RT AC86U?

GerbenGheysens, 2021-06-24:
When I collected some CSI data on the router (4x4) I only got 1 vector in the matlab code. Is there a possibility to distinguish which data is coming from which antenna?

zeroby0, 2021-06-24:
https://github.com/seemoo-lab/nexmon_csi#analyzing-the-csi

> The next two bytes contain core and spatial stream number where the lowest three bits indicate the core and the next three bits the spatial stream number, e.g. 0x0019 (0b00011001) means core 0 and spatial stream 3.
 
Maybe you can use this? 

Also please check if all the 4 antennas are indeed 5 GHz. I suspect 2 of them are 2.4 GHz, and 2 are 5 GHz. Jet - @yujianyuanhaha, opened up the router, and has a better idea. Here is the thread: https://github.com/seemoo-lab/nexmon_csi/issues/199

GerbenGheysens, 2021-06-25:
Thanks for your response, with the provide matlab code it looks like there is just one vector provided. Do you know how I can get the splitted values (the vector from each antenna) with the matlab code?

zeroby0, 2021-06-27:
Packets from all the antennas are in one file, and are read into one vector. Also, you have to enable csi collection for all antennas using makecsiparams. I don't think the matlab script provides ways to distinguish the antennas though. Checkout csikit by @Gi-z.

yujianyuanhaha, 2021-06-28:
@GerbenGheysens

It is pretty easy, in matlab it is like

payloadbytes = typecast(payload,'uint8');
if strcmp(DEVICE,'ASUS')
    
    core = bitand(payloadbytes(56),3); %
    rxss = bitand(bitshift(payloadbytes(56),-3),3);
    Nss(k) = core*4 + rxss;
    
else
    % payloadbytes(55)
    % 0 8 16 24
   % for NEXUS phone and PI
    Nss(k) =  round( payloadbytes(55)/8);
end
right after line 36 at csireader.m

Best

Jet

ozawali, 2021-06-29:
How to setup at the TX and RX router for collected some CSI data on the router (4x4) on ASUS RT AC86U @yujianyuanhaha @zeroby0 Thanks

yujianyuanhaha, 2021-06-29:
@ozawali command like ` ./makecsiparams -c 100/80 -C 0xf -N 0xf -m MAC_ADDR -b 0x88` is good to go.

ozawali, 2021-06-30:
> @GerbenGheysens
>
> It is pretty easy, in matlab it is like
>
> ```
> payloadbytes = typecast(payload,'uint8');
> if strcmp(DEVICE,'ASUS')
>
>     core = bitand(payloadbytes(56),3); %
>     rxss = [...]

Thanks. I use your code, but the rxcore is always zeros.

ozawali, 2021-06-30:
@yujianyuanhaha Thanks. I use your code, but the rxcore is always zeros.

zeroby0, 2021-06-30:
Adding to this, you might wanna not use -b 0x88, that flag allows csi of only packets with 0x88 in frame first byte. Removing it will get you more packets.

GerbenGheysens, 2021-07-05:
Has anyone already managed to get the separate CSI-data vectors from each antenna?

haochenku, 2021-07-12:
Same question here. @GerbenGheysens @ozawali @zeroby0 , I found it is difficult to distinguish which core 1 packet is coming with which core 0 packet corresponding to the same frame. Any Ideas?

GerbenGheysens, 2021-07-29:
I want to do some AoA on 2.4GHz with the router and my iPhone. Is it correct to set the parameters as follows: -C 15 -N1 -m(MAC address) I already found out that the internal antenna is only for 5GHz so this will give all zero's and I've placed the antennas closer to each other (half wavelength)

Gi-z, 2021-07-29:
Was on holiday when this discussion was active, but I've just pushed [an update](https://github.com/Gi-z/CSIKit/releases/tag/ff8234f) (and a subsequent one) to CSIKit which collates CSI for the same frame from multiple spatial streams into one matrix. The implementation is a little work-in-progress in how it is relayed to the user since the real core/spatial stream functionality is not identical to that of the likes of Intel/Atheros CSI tools where dimensions refer to Rx/Tx antennas (As far as I'm aware! If you know better, I'd love to hear more!). However, it does function in its current state. For more detail on the implementation, this [source file](https://github.com/Gi-z/CSIKit/blob/master/CSIKit/reader/readers/read_pcap.py) is loosely based on the MATLAB csireader implementation provided by Seemoo.

For reference for anyone implementing this themselves: the "sequence number" in the payload header will indicate the frame for which CSI was captured. Note that when multiple cores/spatial streams are active, this number will remain the same across a number of frames synchronous with your number of active cores * spatial streams. The core/spatial stream bitmask in bytes 12 and 13 will detail which core/spatial stream the frame was received on.

Any feedback is appreciated, as I'd like to make sure we've all got this worked out.

nisarnabeel, 2022-09-22:
Hi Anyone knows how to change regulations file in asus patch, I am only able to listen to channel 36 at 5GHz?