seemoo-lab/nexmon_csi #228: how to separate the the stream data on each antenna

ozawali, 2021-06-30:
Hi, everyone, I have configured the 4*4 MIMO, but I don't know how to separate those datas. I most concern is that how to know the core of each frame? The device I used is AUSA . Thanks.

zeroby0, 2021-06-30:
The packet layout is described in the readme. It has bits that indicate the antenna and core number. I think it will be faster for you to read it and write a program to separate them than to wait for someone else to do it for you. The matlab code might come in handy for reference.

ozawali, 2021-06-30:
I have did that but the the core and rxss is always zeros. @zeroby0

quantumhub, 2021-07-08:
@ozawali , you can try makecsiparams -c 157/80 -C 2 -N 2 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88
And check if you can get core and txss other than 0s.

haochenku, 2021-07-09:
Hi @zeroby0 @quantumhub , here in 4x4 MIMO case, let say we use -N 2, then AP will automatically send data over 2 streams? This decision is made at STA side?

haochenku, 2021-07-10:
Hi @quantumhub @zeroby0 , will the CSI from the same frame on different stream share the same time or same sequence number?