seemoo-lab/nexmon_csi #229: Understanding the CSI extracted.

haochenku, 2021-07-06:
Dear all,
    I have extracted the CSI out with command 157/80. However, I have some questions related to the extracted CSI. By 157/80, my understanding is that Nexus phone will monitor 80MHz channel containing band 157. But from the 802.11ac channels, 157 is only 20MHz. So will the Nexus phone monitoring band 155 which is 80MHz including band 157, or the Nexus phone will monitor band 157 , 161, 165 ? 

[image]
 
From my results, it is hard to explain why primary channel CSI is at the left side from 1 to 52. I was expecting it should from 106 to 156 something, ie.., the third band. 
[image]

quantumhub, 2021-07-08:
Due to 802.11 channelization, 157/80 actually covers 4x 20MHz channels. They are 149 153 157 161.
From you amplitude vs. subcarrier plot, the 20MHz packets are from 149. It is either from your AP which chooses 149 as its primary channel, or from your neighbor's wifi network which works on 149. It can't be the packet from 157, which should be on subcarriers between approximately 100-150.

haochenku, 2021-07-09:
Thank you. The issue here is I forgot to do fftshift. after fftshift it make senses now. Thanks a lot.

closed 2021-07-09 by haochenku