seemoo-lab/nexmon_csi #234: Use SMA terminator on the receiver， the CSI can still be measured？

Cyrbaby, 2021-07-24:
Thank you for this amazing CSI tool! I have tried to use Matching load  on the core 0 and core 3 on the receiver. Why can I sitll measure the CSI on this two core？ I am very confused about this.
[image]
[image]

matthiasseemoo, 2021-07-24:
The isolation between receive chains is not perfect and the receiver is quite sensitive so most likely you observe the signals received over the antennas of the other receive paths but with a much lower amplitude.

yujianyuanhaha, 2021-07-26:
@Cyrbaby the sma terminator simply cannot perfectly block core 0 3 as you planned ...

haochenku, 2021-07-26:
@Cyrbaby  How do you get this clean PDP?  Thank you.