seemoo-lab/nexmon_csi #235: Zero CSIs from NSS Ant 2 and 3

haochenku, 2021-07-26:
Dear all,
    By following the steps, I configured the NSS to be 2 and 3 by command makeparameters -N 3, 7. I am using Nexus 6P with 2 cores. I am expecting to receive 2 x 3 CSI matrix. However, I can only receive valid CSI on 1st Tx antenna, ie., 2x1 CSI. For other CSI readings, they are all zeros. Is there any clue to solve this?

yujianyuanhaha, 2021-08-12:
@haochenku  we need to setup the TX a bit, i.e. set `iperf3` streaming between a router and another router or a laptop, so as to trigger higher throughput streaming, so you can get 4TX-by-2RX spatial streams.

Best

Jet

yangtat, 2021-08-24:
#144 I've had this problem before, but it solved when I got my router working on 802.11/n.
It is also important to ping the router from another device to improve throughput.