seemoo-lab/nexmon_csi #238: Regarding Carrier Frequency Offset value

elmehdiilli, 2021-08-14:
Hello Guys,

I am noob in Raspberry Pi. I have just purchased one to perform some communication protocols.

I would like to ask you something: I know that the carrier frequency offset is corrected at each frame received. I would like to know if there is a way to get this actual value of frequency offset that was corrected/compensated?

I would genuinely appreciate your help and inputs in this regard.

Thanks so much.

matthiasseemoo, 2021-08-14:
If I remember correctly, there were some registers containing something related to the CFO.

elmehdiilli, 2021-08-15:
@matthiasseemoo Hello Matthias,

Thanks so much for your reply. I would really appreciate your help in this regard. I need to figure out precisely from which register we can get this value of corrected CFO (how much kHz have been compensated?) .

Thank you very much once again,

zeroby0, 2021-08-15:
@elmehdiilli 
Skim through the paper and the phd thesis mentioned in readme, maybe they have details :)
https://github.com/seemoo-lab/nexmon_csi#extract-from-our-license

matthiasseemoo, 2021-08-15:
There are no more details on cfo in my thesis.i wanted to use it as a covert channel, however, never did the implementation on commodity hw.  robin and Francesco looked into cfo correction in the past. maybe it is part of: https://tubiblio.ulb.tu-darmstadt.de/107479/

vlsireddy, 2023-09-26:
Hi,
   so there need to be [CFO, Phase Offset, Sampling Time Offet] correction further after obtaining IQ samples ?

Thanks