seemoo-lab/nexmon_csi #241: bash: makecsiparams: command not found

epuschel, 2021-08-23:
Hardware: CanaKit Raspberry Pi 4

I've followed all of the instructions for the getting started bcm43455c0 guide and have read and followed the instructions on #13 , #19 , and 67 (which for some reason GitHub doesn't want me to link but here it is: https://github.com/seemoo-lab/nexmon/issues/67) but still every time I type "makecsiparams -h" it gives me the same response
`bash: makecsiparams: command not found`

zeroby0, 2021-08-23:
It's not given in the guide, but you would need to build it with `make` on the folder.

I have a script for automating build and documentation for running it here: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51

epuschel, 2021-08-24:
> It's not given in the guide (and I don't know why), but you would need to build it with `make` on the utils/makecsiparams folder, and then link it to somewhere in your $PATH.
> 
> I have a script for automating build, and documentation for running it here: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51 . It also contains some patches to fix a bug which makes nexmon_csi stop after collecting csi for some time.
> 
> Also make sure you're not using the latest version of raspbian, it has kernel 5.10, so nexmon_csi wont work. Details in the link above.

So I've followed your guide but when I try doing 'mcp -h' it gives me the error 'bash: mcp: command not found'.

zeroby0, 2021-08-25:
That's funny! 😂 

Anyhoo, if you ran the script install.sh, it should `make` makecsiparams and link it to somewhere in your `$PATH`.
https://github.com/nexmonster/nexmon_csi/blob/529a3d7852ae4846b74c2a813432914371450f44/install.sh#L52
```bash
setStatus "Installing makecsiparams"
cd utils/makecsiparams
make
ln -s $PWD/makecsiparams /usr/local/bin/mcp
```
Are you following the guide _exactly_ or did you make any changes?

Mweme, 2021-08-30:
Hello Aravind and team, 
I real appreciate much for the solutions you're providing here.
However, I face a challenge once more as when I try to run the command 
**root@raspberrypi:~# nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==**

I just get the following error response.
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

Any viable solution will be much appreciated

zeroby0, 2021-08-30:
Hey @Mweme 

https://github.com/seemoo-lab/nexmon_csi/issues?q=93%3A+Protocol+not+supported

Most of the questions are already answered in the issues, so going through them will save you some time and effort :)

I also have a guide that has been successfully tested by lots of users: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51
If you use the correct version of raspbian os, and build kernel headers, it should work.
> Follow the guide _exactly_ the first time, and then make changes once you get it working once :D

epuschel, 2021-08-30:
I booted up my Raspberry Pi to try out what you said and now my raspberry pi isn't connecting to the internet and it is claiming under network options -> N2 Wireless LAN 'No wireless interface found'.

zeroby0, 2021-08-30:
Yup, that's correct!

It's because headers of the network packets are replaced by their CSI. More details in the paper linked in Readme :)

I use an Ethernet cable to give it network access: https://github.com/seemoo-lab/nexmon_csi/issues/42

https://github.com/seemoo-lab/nexmon_csi/issues/164

epuschel, 2021-08-30:
Gotcha, thanks for the information. I'm going to try following your steps again and I'll let you know if I get this bad boi running.

zeroby0, 2021-08-30:
All the best :)
Let me know if you need any help :)

epuschel, 2021-08-30:
So I redid the installation and everything seemed to run fine, but again I was faced with the same error of mcp not being a valid command. So I went into the directory /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi and noticed that the folder utils was not there, which as you stated earlier your script should have installed.

[image]

epuschel, 2021-08-30:
So I redid the installation which worked and the utils/makecsiparams folder is in the directory There is also the symbolic link between mcp and makecsiparams, but still it's giving me the error `bash: mcp: command not found`

zeroby0, 2021-08-31:
Pretty strange. Can you do `echo $PATH` and see if `/usr/local/bin` is in the path? `mcp` should be in `/usr/local/bin`.

You can also just use the full path to the makecsiparams binary instead of mcp.

epuschel, 2021-08-31:
I did `echo $PATH` and I got `/usr/local/sbin:/usr/sbin:/usr/bin:/sbin:/bin`. I'll try using the full path to the makecsiparams binary.

zeroby0, 2021-09-01:
Ah, superuser's path should be `/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`. It probably got modified on your system somehow?

You can set it like this:
```bash
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

# persist across reboots
echo "PATH=\"/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin\"" >> /etc/environment
```

Also you might wanna check `/etc/profile` to see why the path got modified :)