seemoo-lab/nexmon_csi #243: csi Extraction

Mweme, 2021-09-01:
Hello Team @zeroby0 and others.
i'm currently extracting CSI signal basing on the good contributions you provide here, but when I type the command sudo tcpdump -i wlan0 dst host 192.168.1.6 -vv -w data10.pcap -c 1000 it doesn't provide any response.

The last time I typed sudo ifconfig wlan0 up and I received the message SIOCSIFFLAGS: Operation not possible due to RF-kill
I tried to run the command sudo tcpdump -D, I get the following response
1.eth0 [Up, Running]
2.any (Pseudo-device that captures on all interfaces) [Up, Running]
3.lo [Up, Running, Loopback]
4.bluetooth0 (Bluetooth adapter number 0)
5.nflog (Linux netfilter log (NFLOG) interface)
6.nfqueue (Linux netfilter queue (NFQUEUE) interface)
7.usbmon1 (USB bus number 1)
However, the configuration procedures and installation was made successfully.

Any assistance will be highly recognized.

zeroby0, 2021-09-01:
Yup, rfkill blocks wifi and bluetooth if you don't set WiFi country.

https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51#prepare-raspberry-pi

Also, you need to run tcpdump on the Pi, so I'm not sure what the ` host 192.168.1.6` is.

Mweme, 2021-09-01:
> Heyo, rfkill blocks wifi and bluetooth if you don't set WiFi country.
> 
> https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51#prepare-raspberry-pi
> 
> Also, you need to run tcpdump on the Pi, so I'm not sure what the ` host 192.168.1.6` is.

Hello Aravind, thank you for quick response;
Probably I don't understand  what do you mean running tcpdump on pi, but the command **sudo tcpdump -i wlan0 dst host 192.168.1.6 -vv -w data10.pcap -c 100** was run in the Pi configured with nexmon_csi.
I once more checked the command to **sudo tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000** and it provided the following response

_tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
Got 0_
Your help will be much appreciated.
More over, I typed the command **sudo tcpdump** in the pi and the following are the response came out, do these have something to do with CSI?

09:42:12.770802 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 831559, win 4106, length 0
09:42:12.771079 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 834479, win 4106, length 0
09:42:12.771313 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 837399, win 4106, length 0
09:42:12.771544 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 839183, win 4106, length 0
09:42:12.801943 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 839183:840643, ack 350, win 638, length 1460
09:42:12.802168 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 840643:842103, ack 350, win 638, length 1460
09:42:12.802277 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 842103:843563, ack 350, win 638, length 1460
09:42:12.802353 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 843563:845023, ack 350, win 638, length 1460
09:42:12.802422 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 845023:846483, ack 350, win 638, length 1460
09:42:12.802491 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 846483:847943, ack 350, win 638, length 1460
09:42:12.802559 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 847943:849403, ack 350, win 638, length 1460
09:42:12.802625 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 849403:850863, ack 350, win 638, length 1460
09:42:12.802692 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 850863:852323, ack 350, win 638, length 1460
09:42:12.802759 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [P.], seq 852323:853783, ack 350, win 638, length 1460
09:42:12.802881 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [.], seq 853783:855243, ack 350, win 638, length 1460
09:42:12.802957 IP 192.168.1.6.5900 > 192.168.1.5.54526: Flags [P.], seq 855243:855567, ack 350, win 638, length 324
09:42:12.803419 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 846483, win 4106, length 0
09:42:12.803572 IP 192.168.1.5.54526 > 192.168.1.6.5900: Flags [.], ack 847943, win 4106, length 0

zeroby0, 2021-09-01:
`sudo tcpdump` without any flags captures **all** the packets going through your system, so they are probably just packets from the SSH connection.

It should be fine as long as you're running tcpdump on the Pi. You don't need to use `host 192.168.1.6` though. 

CSI samples only come from UDP port 5000 via wlan0, so running `tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000` should save the CSI samples in a pcap file. Provided you started Nexmon_csi, ofcourse.

Jcelestine, 2021-09-02:
> `sudo tcpdump` without any flags captures **all** the packets going through your system, so they are probably just packets from the SSH connection.
>
> It should be fine as long as you're running [...]

Thank you aravind, however still face challenges, when I run **tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000** I just receive no packet, just the following message **tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
Got 0** is there anyhow this can be fixed?

RafaelsNeurons, 2021-09-15:
I got the same problem like Jcelestine

I run it on a Raspberry Pi 3B+ with 4.19.97 Kernel.
I think the settings are right

@Jcelestine did you fixed it ?

[image]

zeroby0, 2021-09-15:
@RafaelsNeurons 
https://github.com/seemoo-lab/nexmon_csi/issues?q=93 All the best :)

Jcelestine, 2021-09-17:
Yeah I did,

Do the following:
1. Install the compatible nexmon_CSI which is in
https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51#prepare-raspberry-pi
2. Follow the link  https://github.com/seemoo-lab/nexmon_csi/issues?q=93
3. If things didn't work reinstall and start compiling

a)mcp -c 157/80 -C 1 -N 1

b)mcp-h to see the available possibilities

c)*ifconfig wlan0 up *, if wont work start with *sudo ifconfig wlan0 up*

d)nexutil -Iwlan0 -s500 -b -l34 - (keep the output obtained in a) that
start with *CXABEQ*

e)sudo ip link set mon0 up

f) sudo tcpdump -i wlan0

g) sudo tcpdump -wlan0 dst port 5500 -vv -w Rafael.pcap -c 2000

After this, restart the nexmon receiver device, and re-run the command
once more, while running, generate traffic to the transmitter so that
you can visualize the traffic of extracted CSI

All the best,

*Justus M Selestine*
***@***.*** ***@***.***>*
***@***.***
*SkypeId:Justus Selestine*
*Mobile:+821059275989*