seemoo-lab/nexmon_csi #245: No definitions.mk when execute "make install-firmware"

Saberfish, 2021-09-08:
Device: raspberrypi 4B, 
In step 10, when I execute command : "make", it shows :
Makefile:3: /firmwares/bcm43455c0/7_45_189/definitions.mk: No such file or directory
make: *** No rule to make target '/firmwares/bcm43455c0/7_45_189/definitions.mk'.  Stop.

So how to deal with this?

zeroby0, 2021-09-09:
Env variables are not in shell, I think.

Try `source setup_env.sh`

Or use my script to build the firmware and automate loading it into the kernel on reboot: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51

Ps: You have to reload the compiled firmware into your kernel everytime you reboot the Pi.

Saberfish, 2021-09-09:
Thank you for your advice and your script. I've succeed install the firmware.
But when I try to execute "nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=="
it shows:
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)

so I just reboot pi. Then I can't find wireless interfaces. I don't know how to deal with it.
When I execute "ifconfig wlan0 up", it shows:
wlan0: ERROR while getting interface flags: No such device.

And there is another question, howto reload the conpiled firmware into the kernel? By execute the install.sh again?
Thank you very much for your advice

sangeun-jo, 2023-06-02:
You should reload firmware every new terminal session. 
Here is code to reload driver.
```
rmmod brcmfmac 
[Nexmon root directory]/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_[linux kernel version]-nexmon/brcmfmac.ko 
```