seemoo-lab/nexmon_csi #246: Problem with Amplitude of CSI （Nexus5）

chulin5323, 2021-09-10:
Why is the amplitude of some sub-carriers in the 80Mhz CSI data I collected is very large(sub-carriers 0 -60), but the amplitude of most of the sub-carriers is very small?
Equipment used: nexus5
Parameters: makecsiparams -c 157/80 -C 1 -N 1 -m 58:41:20:94:4d:a7
This is the CSI data I collected. 
[image]
[image]

Most of the collected CSI data is shown in the first picture, and only a small part(just a few packages) is shown in the second picture.
Can someone help me? thanks! :)

jlinktu, 2021-09-10:
Hi,
from the [FAQ](https://github.com/seemoo-lab/nexmon_csi#frequently-asked-questions):
> If half or three quarters of extracted CSI for 40 or 80MHz are mostly empty or contain very low values it is very likely that you received a 20MHz frame inside a 40 or 80MHz wide channel. As 40 and 80MHz wide channels are just several bonded 20MHz channels this is totally possible and valid. One of the bonded 20MHz channels will be used as control channel to transmit control frames. Hence, you probably captured CSI of control frames or the transmitter is simply not using the available bandwidth.

Thus, most of the CSI you captured are from 20MHz transmissions.

matthiasseemoo, 2021-09-11:
> Hi @jlinktu <https://github.com/jlinktu> ,thank you very much for your reply! I still have some questions:
> 1.I use my computer to ping the router. If I want to get the ping packet, how to set the -b parameter of makeparameters?
> 2.As you said, I received a 20MHz frame inside a 40 or 80MHz wide channel, how can I solve this problem?
> Thanks in advance for reply.

Best would be to explicitly inject 80 MHz frames.

chulin5323, 2021-09-12:
@matthiasseemoo Thank you very much for your reply. Can I use the app developed by your team to implement frame injection? (E.g. https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_app)
Does this mean that I need two nexus 5 phones, one of which is the sender and the other is the receiver?

matthiasseemoo, 2021-09-12:
One Nexus 5 is sufficient for injection. But you can also do it with any other Nexmon supported chip.