seemoo-lab/nexmon_csi #254: I can't get CSI from a specific mac address

liyenjie33, 2021-12-02:
@zeroby0
Hi! I have installed nexmon_csi on RPi4 and Asus router according to your slide, and it went smoothly when capturing CSI.
But when I wanted to use the modem in my lab as a specific crawl object, I couldn't find the correct channel to capture its CSI.
I want to know how everyone finds the right channel to fetch information.
[image]
[image]
I set the modem at 6/20 and use ac:f1:df:14:91:0e as the address. Then I used `mcp -c 6/20 -C 1 -N 1 -m ac:f1:df:14:91:0e` to generates the parameters.
I first used `mcp -c 6/20 -C 1 -N 1` to capture the CSI without restricting the address, and captured the address from the captured CSI as a setting. It works and captured CSI data, but I don't know where is the address. That is why I want to use the device that I know as a target. 
I would like to ask you if I have set something wrong. looking forward to your reply. Thanks.

liyenjie33, 2021-12-02:
@zeroby0
After several checks, I found that I didn't change channel and bandwidth successfully.
I first use `/jffs/mcp -c 36/80 -C 1 -N 1` to extract CSI.
Then I change to use `/jffs/mcp -c 7/20 -C 1 -N 1`, but I found what I extract are with the same chanspex hash.

I follow this slide to install nexmon_csi on Asus RT-AC86U.
https://github.com/nexmonster/nexmon_csi/discussions/2
Should I change some settings? Maybe I should follow another slide? 
https://github.com/nexmonster/nexmon_csi/discussions/7

When I use `nexutil -k`, `__nex_driver_io: error ret=-1 errno=19` this error shows.
Have you met before?  Looking forward to your reply. Thanks.

liyenjie33, 2021-12-02:
@zeroby0
Hi! I found that `nexutil -Ieth6 -k` can work, and successfully change chanspec from `36/80` to `157/80`.
But if I use `nexutil -Ieth6 -k7/20`, `__nex_driver_io: error ret=-1 errno=22` this error shows.
Is it because this combination is not set in the regulation.c file? Or is nexmon only able to extract CSI at 5GHz?
Looking forward to your reply. Thanks.

zeroby0, 2021-12-02:
Hey @liyenjie33 

> Should I change some settings? Maybe I should follow another slide?
> nexmonster#7

No, you don't have to compile Nexmon_csi yourself unless you need change the source code

> I couldn't find the correct channel to capture its CSI.
> I want to know how everyone finds the right channel to fetch information.

You can either use the router's settings page, or run 
`sudo iwlist wlan0 scan` which scans nearby access points.

> Then I change to use /jffs/mcp -c 7/20 -C 1 -N 1, but I found what I extract are with the same chanspec hash.

wow `/jffs/mcp -c 36/80 -C 1 -N 1`, and `/jffs/mcp -c 7/20 -C 1 -N 1` had the exact same chanspec? That's pretty unexpected. But only a few characters change in the chanspec, so maybe you didn't notice that?

----

`__nex_driver_io: error ret=-1 errno=22`: I don't know what error 22 means. Or error 19.

The `7/20` combination is in the regulations file: https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c#L197, but I don't know why nexmon crashed.

And I have never used nexmon in 2.4 Ghz band, so I don't know how stable it is, but people have succesfully collected CSI in 2.4 Ghz.

But in the 5 GHz band, some broadcomm chips have this bug where only channel 36 is reliable. Maybe that extends to 2.4 too, try all the combinations in regulations.c, restart when nexmon crashes, and find out which combinations work?

liyenjie33, 2021-12-02:
@zeroby0 Thanks for your reply!
I think `/jffs/mcp -c 36/80 -C 1 -N 1`, and `/jffs/mcp -c 7/20 -C 1 -N 1` had the same chanspec is because `7/20` setting will fail? Just like when I use `nexutil -Ieth6 -k7/20`, the error happened and the setting failed. So it will use the previous setting and run out the same result?

Then I will try to see which combinations will cause errors according to your suggestions! I'll tell you if there are any results. I am really thankful for prompt reply!

jlinktu, 2021-12-02:
The RT-AC86U has two separate interfaces for 2.4 and 5Ghz.
Use "eth5" for 2.4GHz and "eth6" for 5Ghz.

liyenjie33, 2021-12-02:
@jlinktu Thanks for your reply!
I will try it!!! You're awesome!!!