seemoo-lab/nexmon_csi #258: problem with nexutil

mamdouhJ, 2021-12-30:
Im using RaspberryPi 4 model b with kernel 5.4.51-v7l+. I went through the tutorial for setting up and when I get to nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA== 
I get this error: 
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received

Im not sure what the problem is.

mraddan, 2023-02-20:
I also face that problem, try to reinstall the firmware. When rebooting theraspberry, the firmware goback to default. It works well for me

Prim1003, 2023-03-15:
> Im using RaspberryPi 4 model b with kernel 5.4.51-v7l+. I went through the tutorial for setting up and when I get to nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA== [...]

Hi, I am currently facing the same problem. Do you know how to fix it? Please help.

zeroby0, 2023-03-15:
You don't necessarily have to reinstall it.

The drivers are replaced with the original drivers and firmware when you restart, so replacing the original files with nexmon generated files inside the SD card will do the trick.

```bash
cd $NEXDIR/patches/bcm43455c0/7_45_189/nexmon_csi/
cd brcmfmac_5.10.y-nexmon
mv $(modinfo brcmfmac -n) ./brcmfmac.ko.orig
cp ./brcmfmac.ko $(modinfo brcmfmac -n)
depmod -a
```

The install script I made does this by default: https://github.com/nexmonster/nexmon_csi/blob/pi-5.10.92/install.sh

There are precompiled binaries available here if you don't need/want to compile nexmon_csi: https://github.com/nexmonster/nexmon_csi_bin

Prim1003, 2023-03-18:
> You don't necessarily have to reinstall/recompile it.
> 
> The drivers are replaced with the original drivers and firmware when you restart, so replacing the original files with nexmon generated files inside the SD card will do the trick. That will make nexmon_csi permanent (till you reinstall raspbian)
> 
> ```shell
> cd $NEXDIR/patches/bcm43455c0/7_45_189/nexmon_csi/
> cd brcmfmac_5.10.y-nexmon
> mv $(modinfo brcmfmac -n) ./brcmfmac.ko.orig
> cp ./brcmfmac.ko $(modinfo brcmfmac -n)
> depmod -a
> ```
> 
> The install script I made does this by default: https://github.com/nexmonster/nexmon_csi/blob/pi-5.10.92/install.sh
> 
> There are precompiled binaries available here if you don't need/want to compile nexmon_csi: https://github.com/nexmonster/nexmon_csi_bin

I tried compiling nexmon_csi_bin but this error occurred:

--2023-03-18 18:34:11--  https://github.com/nexmonster/nexmon_csi_bin/raw/main/base/6.1.19-v8+.tar.xz
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-03-18 18:34:12 ERROR 404: Not Found.
Pre-compiled binaries probably don't exist for your kernel's version: 6.1.19-v8+.
Please create a new Issue on Github and tell us what kernel you are using.

zeroby0, 2023-03-18:
@Prim1003 Yups, you're using kernel 6.1.19-v8+ which Nexmon doesn't support. It's also 64 bit, which nexmon doesn't support either.

Try one of the RaspberryPi OS versions linked in the readme: https://github.com/nexmonster/nexmon_csi_bin#install

Try the 5.10.92

Prim1003, 2023-03-20:
> 
Thank you so much for your help!

rlajm1203, 2025-05-14:
Hi, I am facing that same problem. I using `5.10.92-v7l+` kernel..
please help me

zeroby0, 2025-05-14:
@rlajm1203 I haven't found any recourse other than rebooting the pi and trying again yet. Maybe unloading and reloading the firmware could work, but rebooting has been the most reliable for me.

rlajm1203, 2025-05-14:
> [@rlajm1203](https://github.com/rlajm1203) if you're sure you have setup and configured everything right, I haven't found any recourse other than rebooting the pi and trying again yet. Maybe unloading and reloading the firmware could work, but rebooting has been the most reliable for me.

Thank you for your Quick Answer..

I'll just try to reboot

rlajm1203, 2025-05-14:
> [@rlajm1203](https://github.com/rlajm1203) if you're sure you have setup and configured everything right, I haven't found any recourse other than rebooting the pi and trying again yet. Maybe unloading and reloading the firmware could work, but rebooting has been the most reliable for me.

Thank you. rebooting is work