seemoo-lab/nexmon_csi #262: CSI in TIme Domain

NewbCooder, 2022-01-24:
Hi all, I am currently working on extracting CSI using ASUS-RT86U. I am able to extract CSI just fine. 
Is anyone aware on how to plot the extracted CSI amplitude and Phase in time-domain in Python.
Basically what I need is plot CSI amplitude and phase from each antenna Tx-Rx pair per packet for all the subcarriers.

zeroby0, 2022-01-24:
> in time-domain in Python.
Do you mean it should animate the result in Python?
 
https://github.com/Gi-z/CSIKit

I have a similar thing, but it's only for Raspberry Pis, for now. But you can use/modify the matplolib code to create a performant animation.
https://github.com/nexmonster/nexmon_csi/tree/feature/python

NewbCooder, 2022-01-24:
Hi @zeroby0 , I have 1Tx 3Rx setup and want to plot CSI of each packet for all subcarriers

zeroby0, 2022-01-24:
It's possible, but there is no end-to-end solution yet. You'd have to put together something that reads and decodes CSI data (Gi-z/CSIKit), and something that plot it (Matplotlib?).

What's a 1Tx 3Rx setup? Is there 1 device doing that has an active radio, and 3 that passively record csi?

NewbCooder, 2022-01-24:
Yes, it's 1 TX antenna and 3 Rx antennas

NewbCooder, 2022-01-24:
Thanks @zeroby0 for your input. I will do it and probably share here. Thanks 😊