seemoo-lab/nexmon_csi #263: Rraspberry Pi 4B kernel 5.10 error occurs at make install-firmware

FDJD314, 2022-02-08:
greetings,
I'm installing firmware on the Rraspberry Pi 4B .When I run `make install-firmware` to compile firmware patch, the following error message appears.
make: *** [Makefile:222: gen/ucode.bin] Error 1
And the log files are as follows:
1.ass.log
_Parser ERROR (file "src/csi.ucode.bcm43455c0.7_45_189.asm", line 3421, col 18):
 jext COND_RX_IFS2, skip+
                 ^
syntax error_
2.error.log
_cat: gen/nexmon.pre: No such file or directory_
3.patch.log
_Hunk #5 FAILED at 3328.
...
Hunk #13 FAILED at 3959.
Hunk #14 FAILED at 3962.
Hunk #15 FAILED at 3965.
Hunk #16 FAILED at 3968.
Hunk #17 FAILED at 3971.
Hunk #18 FAILED at 3974.
Hunk #19 FAILED at 3977.
Hunk #20 FAILED at 3980.
...
Hunk #38 FAILED at 5286.
Hunk #39 FAILED at 8658.
11 out of 39 hunks FAILED -- saving rejects to file src/csi.ucode.bcm43455c0.7_45_189.asm.rej_
I have tried some methods to solve these problems, and I hope I can get help.
Any ideas , thanks in advance.

FDJD314, 2022-02-20:
I finished the installation by reading #109 and #225. Thank the developers. It's really a great tool

FDJD314, 2022-02-20:
I finished the installation by reading #109 and #184. Thank the developers. It's really a great tool

closed 2022-02-20 by FDJD314

chixiangw, 2022-04-21:
hi @FDJD314 , Hope you are doing well!
I also got this issue as you did and I just read the two references you mentioned but still haven't figured it out. Can you give me some more hints on how to solve this? I'm also using Raspberry Pi 4, kernel version 5.10 (I think this version is supported right?), and the error msg is as below:

COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
COMPILING src/version.c => obj/version.o (details: log/compiler.log)
ASSEMBLING UCODE src/csi.ucode.bcm43455c0.7_45_189.asm => gen/ucode.bin
make: *** [Makefile:222: gen/ucode.bin] Error 1

Please lemme know if you need any other information. Thanks!

zeroby0, 2022-04-21:
@chixiangw 

Try the pre-compiled binaries at https://github.com/nexmonster/nexmon_csi_bin/. There is an install script there.

Use [Raspbian 2022-01-08](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/) and **don't** run `apt upgrade` before install.

FDJD314, 2022-04-21:
@chixiangw 
I am a real rookie. I solved the problem by running `apt-get install g++` `apt-get install gcc` `apt install python-is-python2`  before `make install-firmware`. Hope it will help you.

FDJD314, 2022-04-21:
@chixiangw
I am a real rookie. I solved the problem by run `apt-get install g++` `apt-get install gcc` `apt install python-is-python2` before `make install-firmware`. Hope it will help you.

chixiangw, 2022-05-04:
> @chixiangw
> 
> Try the pre-compiled binaries at https://github.com/nexmonster/nexmon_csi_bin/. There is an install script there.
> 
> Use [Raspbian 2022-01-08](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/) and **don't** run `apt upgrade` before install.
> 
> That said, the compiler logs might have more information about why the build failed.

Thank you Aravind! This is really helpful. I finally installed nexmon csi with this binary.

chixiangw, 2022-05-04:
> @chixiangw I am a real rookie. I solved the problem by run `apt-get install g++` `apt-get install gcc` `apt install python-is-python2` before `make install-firmware`. Hope it will help you.

Thank you! I remember that I tried those commands you mentioned but I had no luck haha

RUI126, 2022-05-06:
> @chixiangw I am a real rookie. I solved the problem by run `apt-get install g++` `apt-get install gcc` `apt install python-is-python2` before `make install-firmware`. Hope it will help you.

Thank you for your help.I run this command to solve the above problem

lxxlxxxx, 2022-07-18:
Hey! I wonder to ask why occur Error: kernel configuration is invalid  when make install-firmware

SMuzamilAli, 2023-12-21:
> @chixiangw
>
> Try the pre-compiled binaries at https://github.com/nexmonster/nexmon_csi_bin/. There is an install script there.
>
> Use [Raspbian [...]

@chixiangw 

Thank you for providing the pre-compiled binaries. I tried to run it but get error saying:

2023-12-21 11:24:10 ERROR 404: Not Found.

Pre-compiled binaries probably don't exist for your kernel's version: 6.1.0-rpi7-rpi-v8.
Please create a new Issue on Github and tell us what kernel you are using.

My kernel version is: Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux.

can you please provide me with the solution for this kernel version? Would be grateful.

zeroby0, 2023-12-22:
@SMuzamilAli you're not using [Raspbian 2022-01-08](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/), kernel version 6.x is not supported.