seemoo-lab/nexmon_csi #266: Get 0 packet from a specific mac address

Zilliax-Barry, 2022-03-07:
@zeroby0
Hi! Many thanks for your work!
I have installed nexmon_csi on RPi4 according to your slide, and everything was good when capturing CSI.
But when I tried to use a specific Mac address, my phone for example, then the command is 'mcp -c 36/80 -C 1 -N 1 -m $myMac'. 
Next, after I ran ''tcpdump -i wlan0 dst port 5500'', the''0 packets captured'' appreas for all time.
Do you know why and how can I fix it?

zeroby0, 2022-03-12:
Are there any packets if you don't use the `-m $myMac`?

Zilliax-Barry, 2022-03-12:
There is, but it showed an unknown mac address in the file.

zeroby0, 2022-03-12:
Then I think either the myMac is incorrect -- your phone has multiple interfaces, and each has a different Mac ID. Notably, 2.4 GHz and 5GHz radios have different mac ids. Or there is a typo. 

Or your phone is just "weird". Packets from some devices are not picked up for CSI collection for some reason. What happens if you try a different device?

Zilliax-Barry, 2022-03-16:
> Then I think the myMac is probably incorrect -- your phone has multiple interfaces, and each has a different Mac ID. Notably, 2.4 GHz and 5GHz radios have different mac ids. Or there is a typo.
> 
> Or your phone is just "weird". Packets from some devices are not picked up for CSI collection for some reason. What happens if you try a different device?

Sadly, I have changed the WI-FI environment and also the different devices in the past few days but I still got 0 packet. The phone provides a device WLAN Mac address which I used as the $myMAC. Additionally, I used a laptop WLAN physical address.

I wonder if I used the wrong Mac address of my devices so there is no traffic.

closed 2022-03-21 by Zilliax-Barry