seemoo-lab/nexmon_csi #267: Can not capture CSI with pong packets

wolke1526, 2022-03-31:
Hi
I'm using an AC86U router to capture CSI data, it works fine capturing CSI data of packets sent by `iperf`. However, when I try to ping AP router with Station router, I expect the Monitor can capture the return pong packets sent by AP but can not get anything, How do I solve this?
here is the parameter I used `makecsiparams -c 36/80 -C 1 -N 1 -m <macaddr> -b 0x88`
Any suggestions are appreciated!

zeroby0, 2022-03-31:
Don't use `-b 0x88`. It's a filter and allows only a specific type of frames.
- https://en.wikipedia.org/wiki/802.11_Frame_Types
- https://github.com/seemoo-lab/nexmon_csi/issues?q=-b+0x88
- https://github.com/seemoo-lab/nexmon_csi/issues?q=-b+0x88

wolke1526, 2022-04-01:
> Don't use `-b 0x88`. It's a filter and allows only a specific type of frames.
> 
> * https://en.wikipedia.org/wiki/802.11_Frame_Types
> * https://github.com/seemoo-lab/nexmon_csi/issues?q=-b+0x88
> * [Why use frames starting with 0x88? #33](https://github.com/seemoo-lab/nexmon_csi/issues/33)

Thank you for your response! I've another question if I remove the filter option, I will capture CSI data with the control frame which only a quarter of the subcarriers show CSI data, is there any way I can capture fully 80MHz data with only return ping packets?

yjxb1, 2023-09-28:
> > 不要使用 .它是一个过滤器，只允许特定类型的帧。`-b 0x88`
> > 
> > * https://en.wikipedia.org/wiki/802.11_Frame_Types
> > * https://github.com/seemoo-lab/nexmon_csi/issues?q=-b+0x88
> > * [为什么使用以0x88开头的框架？#33](https://github.com/seemoo-lab/nexmon_csi/issues/33)
> 
> 感谢您的回复！如果我删除过滤器选项，我还有另一个问题，我将使用控制帧捕获 CSI 数据，只有四分之一的子载波显示 CSI 数据，有什么方法可以只用返回 pong 数据包捕获完整的 80MHz 数据？我也尝试了选项，因为我注意到 pong 数据包是控制帧类型，但它不起作用`-b 0xb4`

Hello, I also encountered the same problem as you. How can I check the packet type of the router pong?