seemoo-lab/nexmon_csi #271: Help with real time CSI plot

aveusalex, 2022-04-30:
Hello there! Thanks for this awesome work! 

Well, I was looking into the issues and saw zeroby0 answering a guy asking how to plot csi in real time.

He said about listening in port 5500, but I didn't get his tip. If anyone know how to do that and could detail it to me, or send links teaching how to do it, I wouldn't be more thankful. 

Thanks for attention!

Gi-z, 2022-04-30:
Haven't tried it myself, but this was just recently published. Looks promising.
https://github.com/pkindt/CSEye

aveusalex, 2022-05-01:
> Haven't tried it myself, but this was just recently published. Looks promising.
> 
> https://github.com/pkindt/CSEye

Hey man, thank you so much!!!