seemoo-lab/nexmon_csi #272: Shifted CSI graph

alessialuoni, 2022-05-09:
I tried to use Nexmon to capture the CSI using a Raspberry Pi B3+ (I did it in the 2.4 Ghz and using the 20 Mhz bandwith). However I get that the magnitude peak is at subcarrier 29 instead of subcarrier 0. Which error could I have made?
[image]
Moreover how are data ordered wrt subcarriers? Does the complex numbers corresponds to subcarriers -32, -31, -30, ...., 0, 1, ..., 31 or is the order different?

chixiangw, 2022-06-03:
Hi @alessialuoni , I also have a similar question. 

Below is what I got from a test. I was using Raspberry Pi 4Bs as a transmitter and a receiver. In the below figure, I used channel 1 and the bandwidth was set to 20MHz. I feel like the magnitude peak is weirdly high, and the value of the peak is consistent in one test, no matter how I changed the distance and barrier between the transmitter and receiver. 

[image]

Below is what I got when using channel 124 and bandwidth 40MHz. There's no such a peak and the amplitude values look normal. BUT, I looked at the raw numbers in the CSI packets from this test, and I found the weird "peak" still exists, the max value was ~14k, which is much higher than the average normal amplitude values in this test. I was using the Nexmon CSI Explorer tool. So shortly, this weird peak still exists but just didn't show up in the explorer tool.

[image]

@zeroby0 Do you have any thoughts about this peak and its value? I read through the paper and documentation but couldn't find any information about that... Thank you!

mzakharo, 2022-06-04:
Even if you receive 64 sub-carriers in a vector, only 52 are valid for each 20Mhz of bandwidth.  you have to 'mask-out'/filter unused/pilot sub-carriers as these are either constant or noise. for 40/80Mhz those appear as per 802.11 spec. For 20Mhz, on raspberry pi, the mask is special, and looks like it is hardware specific. Experimentally,  you can determine what it is.

zeroby0, 2022-06-04:
Heyo @chixiangw 

Those must be the Null and Pilot sub-carriers. CSI-Explorer removes them [by default](https://github.com/nexmonster/nexmon_csi/blob/feature/python/utils/python/config.py#L30), and that's why you didn't see them in the plots.

The list of indexes for Null and Pilot sub-carriers in here: https://github.com/nexmonster/nexmon_csi/blob/feature/python/utils/python/decoders/interleaved.py#L33 . I heard they're for cellular co-existence, but I don't really know what Null sub-carriers do.

> For 20Mhz, on raspberry pi, the mask is special, and looks like it is hardware specific. Experimentally, you can determine what it is -- mzakharo

That explains why your plot for 20 MHz still has them -- I got the generic indexes from [802.11ac survival guide](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html), but they are apparently different for Pi.