seemoo-lab/nexmon_csi #273: mon0 no such device error raspberry pi 4 B

MaryaRB, 2022-07-27:
hello, I've been trying to run nexmon csi on raspberry pi 4 b, I tried to use kernals 5.10, 5.4, and 4.19, and everytime i get the same errors, when using:
`iw dev wlan0 interface add mon0 type monitor`

the error is:
`command failed: Operation not supported (-95)`

and when using:
`ip link set mon0 up`

i get:
`Cannot find device "mon0"`

I never managed to get anything when using nexutil or tcpdump, the output is always zero. does anyone know a solution to this? thank you in advance.

zeroby0, 2022-07-27:
Hints at patched driver not being loaded, but I'm not sure. What does `nexutil -k` say? I would also check if `dmesg` mentions "Nexmon".

MaryaRB, 2022-07-28:
> Hints at patched driver not being loaded, but I'm not sure. What does `nexutil -k` say? I would also check if `dmesg` mentions "Nexmon".

here are the outputs:

~ $ nexutil -k
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
chanspec: 0x6863, 85/160

~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.97-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1294 SMP Thu Jan 30 13:21:14 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[    0.000000] pcpu-alloc: s36928 r8192 d24512 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 503104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=E4:5F:01:7D:91:8A vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=ea7d04d6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1721804K/2019328K available (8192K kernel code, 687K rwdata, 2408K rodata, 2048K init, 850K bss, 35380K reserved, 262144K cma-reserved, 1232896K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 688 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28692 entries in 85 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000239] Console: colour dummy device 80x30
[    0.000262] console [tty1] enabled
[    0.000307] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000323] pid_max: default: 32768 minimum: 301
[    0.000595] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000612] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001369] CPU: Testing write buffer coherency: ok
[    0.001775] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002422] Setting up static identity map for 0x200000 - 0x20003c
[    0.002581] rcu: Hierarchical SRCU implementation.
[    0.003444] smp: Bringing up secondary CPUs ...
[    0.004246] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005187] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.006086] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006220] smp: Brought up 1 node, 4 CPUs
[... 233 lines cut ...]
[    5.925702] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    5.979595] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    6.057224] bcmgenet: Skipping UMAC reset
[    6.065076] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    6.159049] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    6.159318] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.208482] bcmgenet fd580000.genet eth0: Link is Down
[   11.368489] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control off
[   11.368511] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.277788] Bluetooth: Core ver 2.22
[   12.277858] NET: Registered protocol family 31
[... cut ...]
[   12.289297] Bluetooth: HCI UART protocol H4 registered
[   12.289392] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.289590] Bluetooth: HCI UART protocol Broadcom registered
[   12.507678] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.507684] Bluetooth: BNEP filters: protocol multicast
[   12.507695] Bluetooth: BNEP socket layer initialized
[   12.539420] Bluetooth: RFCOMM TTY layer initialized
[   12.539433] Bluetooth: RFCOMM socket layer initialized
[   12.539447] Bluetooth: RFCOMM ver 1.11

zeroby0, 2022-07-28:
Yup, that's because patched driver module wasn't loaded. 

Backup the original `brcmfmac.ko` supplied with RaspiOS. `modinfo brcmfmac -n` returns it's location. Then replace it with the brcmfmac.ko file at `patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.10.y-nexmon/brcmfmac.ko` - this is the patched driver module.

Repeat that with the brcmfmac43455-sdio.bin at `/lib/firmware/brcm/brcmfmac43455-sdio.bin`. Replace it with `brcmfmac43455-sdio.bin` built with Nexmon.

Now run `depmod -a`, and reboot.

I have an install script that builds Nexmon_csi from scratch and does this here: https://github.com/nexmonster/nexmon_csi/tree/pi-5.10.92. Pre-compiled binaries and scripts for 4.19, 5.4, and 5.10 are here: https://github.com/nexmonster/nexmon_csi_bin

MaryaRB, 2022-07-28:
> 
 when i try to use `modinfo brcmfmac -n` i get this error:
`modinfo: ERROR: Module brcmfmac not found.`
i tried to use your installing script before but it didn't work because of the same error.

zeroby0, 2022-07-28:
Oh.. Did it never work or did it work for a while and then stopped working?

> modinfo: ERROR: Module brcmfmac not found.

Yup, driver module hasn't been loaded.

I'll try to build with my script again this week and see if I can reproduce the error - although I never seem to be able to.

Meanwhile, you can use the precompiled binaries: https://github.com/nexmonster/nexmon_csi_bin

MaryaRB, 2022-07-28:
> Oh.. Did it never work or did it work for a while and then stopped working?
>
>
> Yup, driver module hasn't been loaded.
>
> I'll try to build with my script again this week and see if I can reproduce the [...]

I reinstalled the kernel and used your scrip and it worked, the protocol errors are gone now, however i still get `Cannot find device "mon0"` when i follow the usage section. I also receive 0 packets when using `sudo tcpdump -i wlan0 dst port 5500 -v`. I have the Pi connected to network via ether net cable, I tried to use channel 7, 36, and 157 with no luck, any advice?