seemoo-lab/nexmon_csi #274: Rt-AC86U extracts CSI amplitude and phase

QUANAAAA, 2022-08-01:
Hello, I try to use ASUS RT-AC86U 80M to extract the amplitude and phase of CSI. The centimeter connects the output stream of the transmitter to the receiver. I find that the phase difference between antenna channels is not stable with time, and its shape is similar to rectangular wave. I don't think it is caused by CFO, SFO, PDD, etc. Some people know what causes this, and is there a solution?
Thank you
[image]
[image]

Huxiiiao, 2023-11-27:
Hello, how do you extract CSI?

QUANAAAA, 2023-11-28:
I am not sure if you are asking me how to get CSI with RTAC86U router, or about the CSI amplitude and phase difference obtained in the above case.

Huxiiiao, 2023-11-28:
I used the UbiLocate tool to obtain the CSI, but I am not sure if the CSI obtained is correct, so I ask which method or tool you used to obtain the CSI from RT-AC86U

Huxiiiao, 2023-11-28:
>  I don't know about UbiLocate tool. We use the nexmon firmware, which is available in the Nexmon code section. We followed the steps on the website to collect CSI.
>
>
>
>
>  ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------

OK，thank you very much

   
   
   
   
  ------------------&amp;amp;nbsp;原始邮件&amp;amp;nbsp;------------------  
  主题:&amp;amp;nbsp;Re: [seemoo-lab/nexmon_csi] Rt-AC86U extracts CSI amplitude and phase (Issue #274)

sie2, 2023-12-19:
You can potentially try our tool [nexmon_csi_gui](https://github.com/ConnectedSystemsLab/nexmon_csi_gui/tree/main) which we used to debug issues with RT-AC86U in real-time.

Huxiiiao, 2023-12-19:
OK，thank you very much

closed 2023-12-19 by jlinktu