seemoo-lab/nexmon_csi #275: Setting rx and tx

Cheon932354, 2022-08-02:
Thank you for your wonderful tools.

I am a person who is studying device free indoor localization.
Therefore, i would like to collect wifi information, create a fingerprint through the collected CSI, and estimate the location.

I succeeded in implementing the tool you provided, and I got curious here.

For example, suppose there is a public place where wifi is implemented.
And there is an RP that wants to estimate the location.
Then should I go to RP with Nexmon_csi and measure CSI?

Or should I put a person on the RP and collect CSI by fixing it to the receiver's location?

I'm confused whether nexmon_csi plays a fixed Rx role.

And if I use the udp port 5500, data can be collected even if I don't ping separately, so do I not have to ping separately?

matthiasseemoo, 2022-08-03:
With nexmon_csi you can only receive CSI information.

NewbCooder, 2022-12-27:
Hi @Cheon932354 nexmon makes you receiver to act as sniffer and it has no communication link, you will always need a device to ping the Tx