seemoo-lab/nexmon_csi #278: Play a youtube video on your laptop, and set makecsiparams `-b 0x88 -m 68:ff:7b:b6:25:14`. That way, the data packets from router to laptop will be captured, and other communication will not be.

zikunliu, 2022-09-18:
Play a youtube video on your laptop, and set makecsiparams `-b 0x88 -m 68:ff:7b:b6:25:14`. That way, the data packets from router to laptop will be captured, and other communication will not be. 

0x88 (0b10001000) is QoS Data frames. You can find a list of frame types here: https://en.wikipedia.org/wiki/802.11_Frame_Types

_Originally posted by @zeroby0 in https://github.com/seemoo-lab/nexmon_csi/issues/138#issuecomment-719936038_

zikunliu, 2022-09-18:
Is there anyway that I can only capture data packets from router to laptop instead of capture all data from router to all other device? Thank you

zeroby0, 2022-09-18:
> Is there anyway that I can only capture data packets from router to laptop instead of capture all data from router to all other device? Thank you

It's possible, it's just not implemented.

https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus has both transmitter and receiver mac addresses, so you can filter only the packets you need in the pcap file.

The way to filter receiver address with makecsiparams directly isn't implemented.

zikunliu, 2022-09-18:
Hi Aravind, thanks for quickly response and resource provided. I will try this one. Thank you again.

closed 2022-09-18 by zikunliu

zeroby0, 2022-09-18:
https://github.com/nexmonster/nexmon_csi/blob/pi-5.4.51-plus/src/csi.ucode.bcm43455c0.7_45_189.patch already has the mechanism to copy Transmitter and Receiver mac ids to the CSI packet. 

Changing https://github.com/nexmonster/nexmon_csi/blob/bf0a2bec37976b1e3c2c64c3286003ffad1d8757/src/csi.ucode.bcm43455c0.7_45_189.patch#L112-L155 to

```diff
3336a3425,3462
> spin:
> 	jext	COND_RX_IFS2, skip+
> 	calls	L1120
> 	calls	L1325
> 	jl	SPR_RXE_FRAMELEN, (6 + 24), spin-
> 	jne	[SHM_CSI_COLLECT], 1, skip+
> 	je	r23, 0x0, skip+
> 	je	[APPLY_PKT_FILTER], 0, nopktfilt+
> 	and	[3,off1], 0x00ff, SPARE1
> 	jne	SPARE1, [PKT_FILTER_BYTE], skip+
> nopktfilt:
> 	je	[N_CMP_SRC_MAC], 0, nomacfilter+
> 	mov	[N_CMP_SRC_MAC], SPARE1
> macloop:
> 	jne	SPARE1, 4, macloop2-
> 	sub SPARE1, SPARE1, 2
> 	jne [5,off1], [CMP_SRC_MAC_2_0], macloop2-
> 	jne [6,off1], [CMP_SRC_MAC_2_1], macloop2-
> 	jne [7,off1], [CMP_SRC_MAC_2_2], macloop2-
> 	jne [8,off1], [CMP_SRC_MAC_3_0], macloop2-
> 	jne [9,off1], [CMP_SRC_MAC_3_1], macloop2-
> 	jne [10,off1], [CMP_SRC_MAC_3_2], macloop2-
> 	jmp nomacfilter-
> macloop2:
> 	jne	SPARE1, 2, skip+
> 	jne [5,off1], [CMP_SRC_MAC_0_0], skip+
> 	jne [6,off1], [CMP_SRC_MAC_0_1], skip+
> 	jne [7,off1], [CMP_SRC_MAC_0_2], skip+
> 	jne [8,off1], [CMP_SRC_MAC_1_0], skip+
> 	jne [9,off1], [CMP_SRC_MAC_1_1], skip+
> 	jne [10,off1], [CMP_SRC_MAC_1_2], skip+
> nomacfilter:
> 	or	[8,off1], 0x0, [SRC_MAC_CACHE_0]
> 	or	[9,off1], 0x0, [SRC_MAC_CACHE_1]
> 	or	[10,off1], 0x0, [SRC_MAC_CACHE_2]
> 	or	[14,off1], 0x0, [SEQ_NUM_CACHE]
> 	mov	1, DUMP_CSI
> skip:
```

**might** do what you need. Most of this is guesswork, inferred from the rest of the patch. The syntax is probably wrong, I'm not sure why there is a `+` or a `-` after jumps to `skip+` or `macloop2-`. I'm also suspicious that the syntax in `> 	sub SPARE1, SPARE1, 2` is wrong. You'll have to test it on a real pi and modify till you get it right.

Here is the pseudo code:
```Python
   ...
@ nopktfilter
    if number_of_macs_to_compare == 0: GOTO nomacfilter
    spare_1 = number_of_macs_to_compare
@ macloop
    if spare_1 != 4: GOTO macloop2
    spare_1 = spare_1 - 2
    if mac_recv != macs_compare_2: GOTO macloop2
    if mac_tran != macs_compare_3: GOTO macloop2
    GOTO @nomacfilter
@ macloop2
    if spare_1 != 2: GOTO skip
    if mac_recv != macs_compare_0: GOTO skip
    if mac_tran != macs_compare_1: GOTO skip
@ nomacfilter
    src_mac_cache = mac_tran
    dump_csi = true # probably means copying CSI to userspace?
@ skip
   # Skips CSI collection
   ...
```
You will be able to set upto two sets of transmitter-receiver macid pairs via the usual makecsiparams -b.
```
mac_recv_1, mac_trans_1
mac_recv_1, mac_trans_1, mac_recv_2, mac_recv_2
```
If you give it 1 or 3 mac ids instead of 2 or 4, csi collection won't happen.

zikunliu, 2022-09-18:
Thanks for the explaintion in detail. It is very helpful and I think this is what I need. I will try this and post it when it is correct.