seemoo-lab/nexmon_csi #280: CSI patern changed after several times of collection on Pi 4

chenti29, 2022-10-30:
I have a Pi 4 with nexmon csi installed, and a second pi 4 sending ping packets to router via 5GHz. The first Pi 4 collect csi and generate graph with my own tool.

1, However the patern changed after several times of collection as shown bellow. Is this problem with my installation or problem of Pi?
2, And the csi from second Pi's mac address is different from those from router's address. Is this general?
3, The CSI collection rate is different from my sending rate. It can get 120 packets per seconds when the second Pi sending 100 packets per second. Does it collect some noise data from air even it have mac address filter?

CSI graph for CSI from second Pi at beginning
[image]
CSI graph for CSI from second Pi after times of collection
[image]
CSI graph for CSI from router
[image]

entranezxc, 2022-11-08:
hi, may I ask a question? I'm currently trying to gather CSI packets from a router to a raspberry PI. The router currently has 2 network state which is 2.4Ghz and 5Ghz, I tried gathering thru 2.4Ghz with channel 1 and 20Mhz but I've got no luck gathering CSI Packets. Currently, the wireless signal has a set password, is this an issue?

zeroby0, 2022-11-08:
It's not an issue. Try channel 36 on 5Ghz. Wifi packets from some routers are "invisible" to the pi for some reason.

entranezxc, 2022-11-08:
@zeroby0 thanks, but what should be the bandwidth? 20/40/80 Mhz? I'm trying to select channel 36 from my router but it keeps reverting to 161 at 80 MHz

zeroby0, 2022-11-08:
Any bandwidth is fine. Channel 161 is okay too, it falls under the [channel 157, 80 Mhz umbrella](https://en.wikipedia.org/wiki/List_of_WLAN_channels). Use `-c 157/80` with makecsiparams.

entranezxc, 2022-11-08:
[image]
[image]

@zeroby0 sorry if I have too many questions, I just need to make this work for our thesis since I'm just a student. Can you check if there's something wrong with the commands I entered? It still doesn't give any CSI packets

chenti29, 2022-11-08:
@zeroby0 May I also ask a question about the sampling rate problem?

I saw you have mentioned in issues before that the sampling rate is decided by the sending devices, and it can even reach 1000 Hz. But in my setting, even a rasberry Pi is ping the router 200 packets per second, the CSI Pi can only get arround 150 packets per seconds at most, and this number would decrease under 100 or even 50 packets per second after several times of collection.

And the pattern seems also be "useless" or "noisy" like the second image.

chenti29, 2022-11-08:
@entranezxc Maybe the "nexutil -Iwlan0 -s500 -b -l34" issue I think.
use "-v" with the message produced by mackcsiparams after it.
Like this: "nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=" (for param from "mackcsiparams -c 36/80 -C 1 -N 1")

zeroby0, 2022-11-08:
https://github.com/seemoo-lab/nexmon_csi/issues/280#issuecomment-1306769581
What does nexutil -k say? Try restarting the pi.

zeroby0, 2022-11-08:
@chenti29 

If you're saving the packets to pcap, maybe the storage device is slow?

I run multiple pings and several youtube videos and other stuffs when I need to make several packets. Packet injection would be *perfect*, but I don't have the time to test or implement that right now.

chenti29, 2022-11-08:
@zeroby0 

Thank you for your reply, I am using a SanDisk Ultra micro sd card with U1 writing speed (10MB/s). I would try different sd card.

When you are collecting data, are you listening to router? I thought the ping devices which is closer to the user myself may have more information than the router. Would that be a problem?

chenti29, 2022-11-08:
@zeroby0 

Yes, I did add a mack filter for my devices because I only want csi data that contains information I need.

Thank you for your reply. I would try different micro sd card tomorrow to see if it fix the problem.

entranezxc, 2022-11-08:
@zeroby0 
[image]
Hello, I've successfully gathered CSI Packets by the methods you suggested along with @chenti29 thanks a lot.

However, this is my first time using the raspbian lite version which just includes terminal. How do I get to see or copy the pcap file in order to be analyzed in MATLAB?

zeroby0, 2022-11-08:
@chenti29 in your graphs at the top, did you remove the null and pilot subcarriers from the CSI?

> The CSI collection rate is different from my sending rate. It can get 120 packets per seconds when the second Pi sending 100 packets per second. Does it collect some noise data from air even it have mac address filter?

There's a lot of traffic going on other than the ping packets you're sending. Even if no devices are connected, router broadcasts packets for discoverability.

chenti29, 2022-11-08:
@zeroby0 

I did not remove the null and pilot subcarriers. I did not find a convincible document that explain the subcarriers' structure.

Those broadcasts packets may be the problem I am facing. Do you know where I can got those related information?

entranezxc, 2022-11-08:
@chenti29 Thanks for the suggestion, I've successfully configured SSH in my PC, now I can control the raspberry pi in my desktop. My question is what comes after the 'scp pi@' in your reply? should it be the ip address of my raspberry pi?

chenti29, 2022-11-08:
@entranezxc yes, and the "pi" before "@" is your Pi's user name, which is "root" according to your image I think.

zeroby0, 2022-11-08:
There's many kinds of broadcast packets, so I don't know which one to point you at, but here is a wiki of packet types I use: https://en.wikipedia.org/wiki/802.11_Frame_Types

Packets from 5.10 version have fctl in them, they might help you identify the frame types to an extent.

I have a list of pilot and null subcarrier indices here: https://github.com/nexmonster/nexmon_csi/blob/feature/python/utils/python/decoders/interleaved.py#L33. 

If you edit config.py, csi-explorer can remove the nulls and pilots. There's also https://github.com/nexmonster/nexcsi to read pcap files, but it doesn't have plotting capabilities yet.

zeroby0, 2022-11-08:
Afaik you can't ssh into pi as root because of it's default ssh config. The image shows root because the nexmon_csi commands have to be run as root.

chenti29, 2022-11-08:
@zeroby0 

Thank you for your help, that help a lot.
I would refer to those informations, it is really helpful.

zeroby0, 2022-11-10:
> @zeroby0 May I also ask a question about the sampling rate problem?
>
> I saw you have mentioned in issues before that the sampling rate is decided by the sending devices, and it can even reach 1000 [...]

How are you generating traffic?

I have a router operating at 80 Mhz, to this router I have a pc running linux connected via gigabit Ethernet, and a laptop running linux via WiFi. The pi is also connected via Ethernet to the router but that's not really relevant.

I start iperf3 server on the pc: `sudo iperf3 -s`, and iperf client on the laptop: `sudo iper3 -c ip-address-of-pc -i 0 -t 30000`. This generates ~350 Mb/s of wireless traffic, and I'm able to collect ~12,000 packets on the pi in 10 seconds.

chenti29, 2022-11-15:
@zeroby0 
Sorry for reply late.
I have changed the microSD card of Nexmon Pi from 10MB/s of writing speed to 90 MB/s, this solved the low sampling rate problems for now. I have collect CSI with 300Hz and it works well after times of collection.

> How are you generating traffic?

I have a router operating at 80 MHz channel 36, a Windows pc connect via Wi-Fi, a Nexmon raspberry pi connect via Ethernet and another pi connect to router via Wi-Fi.

I ssh to both pi, ask second pi to ping the router at some frequence, usually 100 Hz with `sudo ping router_ip_address -i 0.01`, and let the first one to listen to CSI. I would add mac filter withe "-m" param, for now I am only listen from second pi.

At the botton is my new CSI data, it get 266 Hz of data. And one problem left is actually during this image, the second pi only ping 100 packets per second to the router `sudo ping router_ip_address -i 0.01`, and the actuall performence is about 80 Hz. However the first pi collect 266 Hz of data. Are the pi listen to a packet more then one time so it get 3 times of frequence of the expect?

[image]

zeroby0, 2022-11-15:
You can forward the CSI packets to your laptop or some other device and collect them there if the SD card size / speed is a problem (https://github.com/nexmonster/relay).

Reducing the bandwidth to 40 MHz can help too. I measured ~2200 packets per second at 40 MHz with 100 mbps of iperf3 traffic.

> Are the pi listen to a packet more then one time so it get 3 times of frequence of the expect?

When you send a ping, the other device replies. Pi is listening to both the original ping and it's reply, so there are twice the packets. Apart from the traffic you're generating with ping, your devices send a lot of other data. For example, 'heartbeat' packets are sent between device and router to tell that router that the device is still connected. There are announcements in the network when a new device joins the network. Your router periodically sends packets to figure out the best beam forming settings. Etc.

There's a lot of communication going on between your devices even if you're not doing pings or other data. That's where the extra packets come from. Not all of those packets use 80 MHz. Some packets only use 20 MHz to be compatible with all devices. If there is noise, or there isn't a lot of data to transmit, your device might choose to transmit  with only 20 or 40 MHz.

There's lot more discussion and useful information in old issues. I **strongly** recommend skimming through them and then the nexmon_csi paper.

zeroby0, 2022-11-15:
PS: If you have wireshark installed, scan your wifi (menubar > capture > start) and you'll see all kinds of fun packets. Even if you're not creating any data yourself!

chenti29, 2022-11-15:
@zeroby0 

Thank you for your advices, I have force the router to work on channel 36, 80 MHz, and add a mac address filter in mcp command. So I think those noise are "heartbeat" packets or other communications like you said. I tried the iperf3 stuff and it works pretty well right now.

Thank you a lot for your help, I have read many old issues before and I think I should read the paper again.

Since my issue is solved, I think I would close this issue here.

closed 2022-11-15 by chenti29