seemoo-lab/nexmon_csi #284: Used the pkill wpa_supplicants command now Wi-Fi shows no wirless interfaces found  and when follow the steps the following shows up

razer8666, 2022-12-11:
[
[image]
](url)

zeroby0, 2022-12-11:
That's true. You won't be able to use the WiFi chip's usual functions when Nexmon is installed. 

Errors with `nexutil` could be because of a lot of things. One of the reasons is firmware crashing (which may or may not have happened in your case), but restarting the Pi fixes that for me. Make sure you reload the firware after you restart.

With `makecsiparams`, the `-b 0x80` and `-m macaddress1:macaddress2` are included in the readme so that you're aware they exist, but they aren't meant to be copy-pasted.

The `-b 0x80` is a filter that only allows packets whose [Frame Control byte](https://en.wikipedia.org/wiki/802.11_Frame_Types) is `0x80`. If you're trying to capture CSI of all packets, you shouldn't use it.

Similarly, the `-m` filter selects only packets arriving from matching Mac addresses for CSI collection. Since you don't have `00:....:55` or `aa:bb:...:dd` mac ids in your wireless network, you won't see any CSI traffic.

Pre-compiled binaries for nexmon_csi are available here: https://github.com/nexmonster/nexmon_csi_bin, so you won't have to compile Nexmon_csi unless you need to. There's also a few other tools available in the same org: https://github.com/nexmonster. If you install the binaries from `nexmon_csi_bin`, you don't have to reload them manually after restarting.

razer8666, 2022-12-11:
so how do I capture the CSI packets ? Since the wlan0 is not functional, I am not connected to any router, do I need to manually connect it to any router ?.  It would be very helpful if you could list the steps for the usage.

zeroby0, 2022-12-12:
CSI collected works, just the other normal wifi functionality doesn't.

You're using `-c 157/80`, this collects All wifi packets going in the air on channel 157 bandwidth 80. See the 5ghz table here for what frequencies that includes: https://en.m.wikipedia.org/wiki/List_of_WLAN_channels

 It's like monitor mode. You don't need to be connected to anything to listen. Except monitor mode gives you packet contents, and nexmon-csi gives you packet csi.

zeroby0, 2022-12-12:
Ah, so if you want csi, make sure the devices you're interested in are transmitting in the channel you're listening to.

razer8666, 2022-12-13:
Hi i compiled the binaries from the  nexmon_csi_bin but now when i use the ifconfig wlan0 up it shows wlan0: Error while getting interface flags: No such device

zeroby0, 2022-12-13:
Did you do it on a fresh raspbian install and restart after installing nexmon_csi_bin?

Also, restarting might help

razer8666, 2022-12-13:
yes I installed it on a fresh raspbian install and restarting is not helping. Please help

razer8666, 2022-12-13:
I managed to get the wlan0 up but now when I use the nexutil command it shows nexutil : invalid option --'1'

zeroby0, 2022-12-14:
Yeah cause there is no option `-1`.  Did you type `-134` instead of `-l34` (lowercase L).

They seem to look very similar for some reason on github's monospace font, but if you copy-paste them it should be correct.

razer8666, 2022-12-14:
I fixed it I was using 1 instead of using l thanks a lot. When I plot the CSI values the change isnt that significant any suggestions how do I monitor the CSI values more precisely ? ( I am using the CSI values for human activity recognition ).

zeroby0, 2022-12-15:
You should remove the null and pilot subcarriers. They have large and arbitrary values, so actual csi values aren't visible in plots.

I'll post the code to do that when I get to my office in a couple of hours.

razer8666, 2022-12-15:
Okay thanks bro I will be waiting.

zeroby0, 2022-12-15:
```pip3 install nexcsi```

You'll need [nexcsi](https://github.com/nexmonster/nexcsi) version 0.5.2 or above.

```python

import numpy as np
import matplotlib.pyplot as plt

from nexcsi import decoder

device = "raspberrypi" # nexus5, nexus6p, rtac86u

samples = decoder(device).read_pcap('0.pcap') # pcap file path here

class Plotter():
    def __init__(self, samples, device):
        self.samples = samples
        self.device = device

        nfft = int(samples.dtype.metadata['bandwidth'] * 3.2)

        self.x = np.arange(-1 * nfft/2, nfft/2)

        fig, ax = plt.subplots(2, 1)

        self.ax = ax

        plt.ion()
        plt.show()
    
    def get_frame(self, index, zero_nulls, zero_pilots):
        return decoder(self.device).unpack(self.samples['csi'][index], zero_nulls=zero_nulls, zero_pilots=zero_pilots).ravel()
    
    def update_data(self, index, zero_nulls=True, zero_pilots=True):
        plt.cla()

        self.ax[0].cla()
        self.ax[1].cla()

        self.ax[0].title.set_text(f'Channel State Information')

        csi = self.get_frame(index, zero_nulls=zero_nulls, zero_pilots=zero_pilots)

        self.ax[0].plot(self.x, np.abs(csi))
        self.ax[1].plot(self.x, np.angle(csi))

        self.ax[0].set_ylabel('Magnitude')
        self.ax[1].set_ylabel('Phase')
        self.ax[1].set_xlabel('Subcarrier')

        plt.ion()
        plt.draw()
        plt.pause(0.001)

plotter = Plotter(samples, device)
```
#### To plot a single sample

Shows the 101st sample (indexes start from 0 in python)
```python
plotter.update_data(100)
plt.show(block=True)
```

#### Show multiple samples as animation
Plots the first 100 samples, updating the plots every 0.3 seconds
```python
import time

for i in range(100):
    plotter.update_data(i)
    time.sleep(0.3)
```

razer8666, 2022-12-15:
Hi I used the code but for some reason, the graph has no values its empty. I am also attaching a picture for your reference. 
[image]

zeroby0, 2022-12-15:
[0.zip](https://github.com/seemoo-lab/nexmon_csi/files/10236113/0.zip)

Does it work with this pcap?

razer8666, 2022-12-15:
No it doesnt work with this as well.

razer8666, 2022-12-15:
got it working thanks man

closed 2022-12-15 by razer8666