seemoo-lab/nexmon_csi #287: Frame Injection from ASUS RT-AC86U (bcm4366c0)

Davide-Zamboni, 2023-02-03:
I am trying for days to implement frame injection on ASUS RT-AC86U (bcm4366c0) without any success.  I want simply to use two routers, one to send a custom payload on a specific bandwidth and the other will do the CSI extraction.

I read this: [https://github.com/seemoo-lab/nexmon_csi/issues/166](url). It seems someone has already accomplished this but the code isn't public. Furthermore, as stated here: [https://github.com/seemoo-lab/nexmon_csi/issues/16](url), it simply may consist in the implementation of the sendframe.c file for the bcm4366c0, but it requires reverse engineering the firmware.

Any Help?

@zeroby0

zeroby0, 2023-02-08:
I think I remember reading @Gi-z implemented frame injection a while ago, but it was unstable.  And I don't remember if it was for rt-ac86u, it might've been for pi. As far as I understand, sendframe.c has not been implemented for bcm4366c0: https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm4366c0/10_10_122_20

You **might** be able to install something like aircrack-ng on merlin-wrt and inject packets there, but I don't know much about it. I recommend asking in https://github.com/seemoo-lab/nexmon/issues though - code for injection lives in that repository.

jlinktu, 2023-02-09:
Due to the frequent request I will add a frame transmission/injection example patch for the `bcm4366c0` to the [nexmon](https://github.com/seemoo-lab/nexmon) repository in the next few days.

jlinktu, 2023-02-17:
Here you go:
https://github.com/seemoo-lab/nexmon_tx_task

Davide-Zamboni, 2023-02-17:
Thanks a lot @jlinktu 🎉

closed 2023-02-17 by Davide-Zamboni