seemoo-lab/nexmon_csi #289: make install-firmware in nexmon_csi error “recipe for target 'gen/ucode.bin' failed”

Dragon6879, 2023-02-25:
am using the nexus 5  and common router to get csi
but when make install-firmware in /nexmon/patches/bcm4339/6_37_34_43/nexmon_csi#
error:
COLLECTING STATISTICS read /nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  ASSEMBLING UCODE src/csi.ucode.bcm4339.6_37_34_43.asm => gen/ucode.bin
Makefile:218: recipe for target 'gen/ucode.bin' failed
make: *** [gen/ucode.bin] Error 1

is i using the common router? must be ac86u?

zeroby0, 2023-02-25:
No it doesn't need to be an ac86u. 

There should be a log folder somewhere nearby, with files like compiler.log and patch.log, that might have helpful information :)

Also, which version of the kernel are you running on your phone? Is android always kernel 4.19?

Dragon6879, 2023-02-25:
thanks very a lot .i have solved this error.but got a new problem.
when i "tcpdump -i wlan0 -vv dst port 5500 -w /sdcard/csi.pcap"
it shows GET 0, and if i delete the "dst port 5500" it will get package. i don't know why 

my nexus 5 's Andorid is 6.0.1
Kernel version is 3.4.0-gcf10b7e
version number is M4B30Z

zeroby0, 2023-02-25:
What does `nexutil -k` say?

`dst port 5500`  is a filter to capture only the packets nexmon_csi is sending. There's a Lot more packets that your phone sends (the rest of the internet traffic), and those are what's being shown when you remove it.

what parameters are you using for makecsiparams (mcp) and the rest of the commands? If there's no wifi traffic in the channel you're scanning, nothing is going to be reported.

Dragon6879, 2023-02-27:
sorry to reply you too late
In the WiFi settings of the Xiaomi router I used, 
I adjusted the wireless channel to 7 and the band bandwidth to 20M.
so i use the makecsiparams commands,i put "./makecsiparams -c 7/20 -C 1 -N 1 -m 5C:02:14:08:AA:08
"
i see someone add "-b 0x88" after mac address,i don't know how these parameters works.
you said dst port 5500 is a filter, isn't i use the correct port?

Dragon6879, 2023-02-27:
and i use the "nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==" command to Configure CSI extractor. these parameters have some problems?

Dragon6879, 2023-02-27:
i also find if i don't put a specific mac address.
"./makecsiparams -c 7/20 -C 1 -N 1"
i will get data. but if add a specific mac address,it will get none

zeroby0, 2023-02-27:
> i will get data. but if add a specific mac address,it will get none

The mac address you're using is probably incorrect then. 2.4 and 5ghz radios have a different mac id.

> i see someone add "-b 0x88" after mac address,i don't know how these parameters works.

https://www.wikiwand.com/en/802.11_Frame_Types

The first byte of a packet signals it's frame type. If you use `-b 0x88` only packets whose first byte is `0x88` are captured. You can filter for packets of a specific frame type with it.

Yes the port 5500 is correct. You need to use it with tcpdump.

aafsoul, 2023-06-08:
Hello, I have the same problem as you. How did you solve it

Dragon6879, 2023-06-12:
which one？cannot get message?

Dragon6879, 2023-06-12:
> Hello, I have the same problem as you. How did you solve it

which one ?

sarielkk, 2023-08-15:
> 非常感谢，我已经解决了这个错误，但遇到了一个新问题。当我“tcpdump -i wlan0 -vv dst port 5500 -w /sdcard/csi.pcap”时，它显示 GET 0，如果我删除“DST 端口 5500”，它将得到包。我不知道为什么
> 
> 我的连结 5 的 Andorid 是 6.0.1 内核版本是 3.4.0-gcf10b7e 版本号是 M4B30Z

> thanks very a lot .i have solved this error.but got a new problem. when i "tcpdump -i wlan0 -vv dst port 5500 -w /sdcard/csi.pcap" it shows GET 0, and if i delete the "dst port 5500" it will get [...]

May I ask your question, how to solve the problem of nexus5 installation firmware error?

Dragon6879, 2023-08-20:
> 

https://www.cnblogs.com/cjl520/p/17169476.html
我已经忘记如何解决的了，我大概记得是重新下载正确的包和注意版本问题，您可以看一下这个连接，希望对你有帮助

mwfong-csl, 2024-04-29:
When building nexmon-csi for an Asus RT-AC86U (bcm4366c0) under Ubuntu 22.04, `log/ass.log` shows
```
Parser ERROR (file "src/csi.ucode.bcm4366c0.10_10_122_20.asm", line 125, col 9):
 xje [SHM(0x1724)], 0, skipdeaf+
        ^
syntax error
```
When examining  `src/csi.ucode.bcm4366c0.10_10_122_20.asm` against `https://github.com/mbuesch/b43-tools/blob/master/assembler/scanner.l`, I can't find a definition for the opcode, `xje`.  Additionally, `[SHM(0x1724)]` doesn't match `https://github.com/mbuesch/b43-tools/blob/master/assembler/parser.y`'s definition for `mem` -> `imm` -> `immvalue`.  Is `nexmon.git/buildtools/b43-v2/assembler` the correct tool version for this code?

Suggestions?  Thanks!