seemoo-lab/nexmon_csi #291: Fatal error: brcmu_utils.h: No such file or directory

avi3shake, 2023-04-28:
I am trying to build the nexom-csi patch on Rpi4
Kernel : 5.4.83
I have successfully followed all the steps as mentioned in the readme.

`git clone https://github.com/seemoo-lab/nexmon_csi.git`
Cloning is done successfully. However the error pops up while make install-firmware

```
 cd nexmon_csi/
 make install-firmware

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
  COMPILING src/csi_extractor.c => obj/csi_extractor.o (details: log/compiler.log)
  COMPILING src/patch.c => obj/patch.o (details: log/compiler.log)
  COMPILING src/regulations.c => obj/regulations.o (details: log/compiler.log)
  COMPILING src/console.c => obj/console.o (details: log/compiler.log)
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  DISASSEMBLING UCODE /home/pi/nexmon/firmwares/bcm43455c0/7_45_189/ucode.bin => gen/ucode.asm
  PATCHING UCODE src/csi.ucode.bcm43455c0.7_45_189.patch => src/csi.ucode.bcm43455c0.7_45_189.asm
  ASSEMBLING UCODE src/csi.ucode.bcm43455c0.7_45_189.asm => gen/ucode.bin
  COMPRESSING UCODE gen/ucode.bin => gen/ucode_compressed.bin
  GENERATING C FILE gen/ucode_compressed.bin => src/ucode_compressed.c
  COMPILING src/ucode_compressed.c => obj/ucode_compressed.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/ucode_compression_code.c => obj/ucode_compression_code.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/wrapper.c => obj/wrapper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/argprintf.c => obj/argprintf.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/helper.c => obj/helper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/udptunnel.c => obj/udptunnel.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/objmem.c => obj/objmem.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/securitycookie.c => obj/securitycookie.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/radiotap.c => obj/radiotap.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/firmwares/bcm43455c0/7_45_189/flashpatches.c => obj/flashpatches.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  GENERATING LINKER FILE gen/memory.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.c:12:10: fatal error: defs.h: No such file or directory
 #include <defs.h>
          ^~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/chip.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.c:21:10: fatal error: brcmu_utils.h: No such file or directory
 #include <brcmu_utils.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon/cfg80211.o] Error 1
make[1]: *** [Makefile:1732: /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_5.4.y-nexmon] Error 2
make: *** [Makefile:108: brcmfmac.ko] Error 2

```

zeroby0, 2023-04-29:
`source setup_env.sh` wasn't run is my guess

avi3shake, 2023-04-29:
I did run the source setup. I will look into the nexmonster and update here. Thanks

avi3shake, 2023-04-29:
I managed to get the nexom-csi work based on the script mentioned in nexmonster.
However, I am getting the socket error
```
root@raspberrypi2:/home/pi# nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
```

zeroby0, 2023-04-30:
if you installed it on the right kernel version, a restart usually fixes this error for me.

avi3shake, 2023-04-30:
My kernel is Kernel : 5.4.83

avi3shake, 2023-05-03:
Prepare Raspberry Pi
Burn [Raspbian Buster Lite 2020-08-20](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/) onto an empty SD card. You can use [Etcher](https://www.balena.io/etcher/).
[Create an empty file called ssh](https://www.raspberrypi.org/documentation/remote-access/ssh/), without any extension, on the boot partition of the SD card.
[SSH](https://www.raspberrypi.org/documentation/remote-access/ssh/) into the Pi via Ethernet.
With sudo raspi-config, set WiFi Country to US, set your Time Zone, and then Expand File System.
Reboot when asked to.

I am trying to use this instruction, but somehow I cannot ssh into the pi with raspberrypi.local. I am not sure where I am going wrong.

zeroby0, 2023-05-05:
Try using the actual IP instead of raspberrypi.local. You can find it in the router's settings page.

sangeun-jo, 2023-05-30:
I had same problem but solved with 5.10.92+ linux kernel.

rrathnayaka203, 2024-07-25:
> I had same problem but solved with 5.10.92+ linux kernel.

Thank you very much. This saved the day.