seemoo-lab/nexmon_csi #292: CSI collection not working correctly

cheneso89, 2023-05-05:
I tried to collect CSI with ASUS RT-AC86U but this is the plot that I got, that's the CSI amplitude (80 Mhz) for each receiving antenna. I couldn't find what's the problem while I'm collecting it.
[image]

zeroby0, 2023-05-06:
Er, it doesn't seem the right nulls and pilot values have been removed.

For an 80 MHz channel, these are the indices `[0, 1, 2, 3, 4, 5, 25, 53, 89, 117, 127, 128, 129, 139, 167, 203, 231, 251, 252, 253, 254, 255]` that must be made 0. Maybe I'm not looking right, but it seems indices 0->5 are non zero in the graphs.

Or maybe your CSI is just like that. Can you post a pcap with 10-20 packets?

cheneso89, 2023-05-07:
the indices seem correct. This is one of my pcap files

[lab_trial_2.pcap.zip](https://github.com/seemoo-lab/nexmon_csi/files/11414243/lab_trial_2.pcap.zip)

zeroby0, 2023-05-10:
```python
import numpy as np
import matplotlib.pyplot as plt

from nexcsi import decoder

device = "rtac86u" # nexus5, nexus6p, rtac86u
samples = decoder(device).read_pcap('lab_trial_2.pcap')

# Accessing CSI as type complex64
csi = decoder(device).unpack(samples['csi'], zero_nulls=True, zero_pilots=True)

f, (ax1, ax2) = plt.subplots(2, 1)
ax1.plot(np.abs(csi[0])) # Amplitude
ax2.plot(np.angle(csi[0])) # Phase

plt.show()
```

[image]

cheneso89, 2023-05-10:
I already saw this code.. I'm working on Matlab though

zeroby0, 2023-05-11:
Ah. Then maybe modify it to set these indices to 0. Python array index starts from 0 while matlabs starts from 1, so maybe that's the confusion?

I don't know matlab much, however, so I can't help you with that.

cheneso89, 2023-05-11:
No no I already shifted indices by 1, it doesn't work. I'm going to work with python. Thank you anyway

lionlanlan, 2024-06-18:
Hi，sorry about to disturb you. I am also using AC86U to collect data but after collection (tcpdump -i eth6 ) where can I find the file I got from this tool?thx!

jlinktu, 2024-06-18:
Your question is not related to this issue. Please, don't choose random issues and post your questions there!
Also, `tcpdump` is not a tool by us. You can find a very detailed description of its usage with a quick online search, e.g.: [the tcpdump man page](https://www.tcpdump.org/manpages/tcpdump.1.html).

lionlanlan, 2024-06-18:
> Your question is not related to this issue. Please, don't choose random issues and post your questions there! Also, `tcpdump` is not a tool by us. You can find a very detailed description of its [...]

I am very sorry for my impoliteness.I will delete my comment.Sorry for this again.