seemoo-lab/nexmon_csi #299: Missing value of csi file extracted with nexmon extractor

rack570, 2023-06-16:
Hi, I am a nexmon csi extractor user. I found something strange when I extracted the csi and looked at the data from RasPi. When I captured 1000 packets, when I looked at the pcap file with wireshark, I saw 1000 packets correctly, but when I plotted it, I can see a lot of missing values. In fact, when converting a pcap file to a csv file, huge missing values are found in the last stage of extraction. Does anyone know about this problem? Could it be because Raspberry Pi's hard issue or SD card is old? And if you extracted 1000 packets, there should be 1000 rows in csv, but this file only has 944 rows..

[image]

mcp -C 1 -N 1 -c 36/80
tcpdump -i wlan0 dst port 5500 -vv  

these are my pcap, csv files.
[36ch80mhz.zip](https://github.com/seemoo-lab/nexmon_csi/files/11766057/36ch80mhz.zip)

Asafe33, 2024-06-05:
Did you understand the reason for the errors?  What was your solution?

jlinktu, 2024-06-06:
@Asafe33 please stop spamming on every open issue of this repository, if you have a specific problem that was not already solved in an existing issue thread you can create a new issue

Asafe33, 2024-06-06:
Dear @jlinktu ,

I am a physiotherapist and have been working with CSI data capture for just over four months. Due to my limited expertise and the challenges I've encountered during this time, I have found valuable solutions in the repository to the problems I have been facing. However, I have noticed that some issues have been open for 1-2 years, and I would like to understand if any solutions have been found.

Sincerely,

jlinktu, 2024-06-06:
I can understand your interest, still, there is no need to ask about every open issue. If the issue was not closed by us or the opener, the issue was either not relevant to this repository or the owner couldn't find a solution/decided to not post about it.
Everything you should need to know to capture CSI with this project was already handled in one of the issues.
If you encounter a new problem you can open a fresh issue.

Asafe33, 2024-06-06:
@jlinktu Wow, now I get it. Thank you so much for clarifying! 😊