seemoo-lab/nexmon_csi #300: I can't build makecsiparams for Nexus 5

sangeun-jo, 2023-07-04:
If I use `make` command in `/nexmon_csi/utils/makecsiparams`, 
it returned `no rule to make target 'libs/armeabi/makecsiparams'`.

How to build makecsiparams for Nexus 5?

sangeun-jo, 2023-07-12:
I succeeded to build makecsiparams for Nexus 5. 

Android.mk
```LOCAL_PATH := $(call my-dir)

include $(CLEAR_VAR)

LOCAL_SRC_FILES := \
        makecsiparams.c \
        bcmwifi_channels.c
LOCAL_MODULE := makecsiparams
LOCAL_MODULE_FILENAME := makecsiparams

LOCAL_CFLAGS += -std=c99
LOCAL_CFLAGS += -fPIE
LOCAL_LDFLAGS += -fPIE -pie
LOCAL_C_INCLUDES += ./includes 

AL_MODEL_PATH := $(TARGET_OUT_OPTIONAL_EXCUABLES)

include $(BUILD_EXECUTABLE)
```
*Noted: I moved all header files to include folders
```
mkdir includes
cp *.h includes
```
Application.mk
```
APP_FLOATFORM := anroid-20
APP_BUILD_SCRIPT := Android.mk
APP_ABI := armeabi
```

build command 
```ndk-build NDK_APPLICATION_MK=`pwd`/Application.mk NDK_PROJECT_PATH=`pwd` ```

I hope this will be added this repo. 
@jlinktu

closed 2023-07-12 by sangeun-jo

doge0823, 2023-12-31:
Hello, I want to use makecsiparams locally to generate relevant parameters on Android phones. Can these two mk files and build command be applied to Nexus6p phones