seemoo-lab/nexmon_csi #303: The packet capture speed is slow

XXINCH, 2023-07-20:
hi, guys
I'm using mobile hotspot as AP and Raspberry PI 4B for packet capture, but when I run `tcpdump -i wlan0 dst port 5500 -c 1 -w 1.pcap`, it usually takes a long time to catch a packet.
I also tried using a Windows computer to connect to the phone hotspot and ping the phone's ip address from the computer. However, in this case, whether I set the mac address filtering to the mac address of the phone or the mac address of the computer, the packet capture was slow.
Also, the channel of the phone hotspot I use is 149.

XXINCH, 2023-07-20:
@zeroby0

zeroby0, 2023-07-20:
Try running a fast.com speed test on the computer. That should generate lots of wireless traffic. Try with no mac filtering first, and then add filtering to see if the problem is the wrong mac address :)

XXINCH, 2023-07-20:
@zeroby0 Thanks for your reply. Since I cannot run the speed test, I tried other ways like playing some videos to generate wireless traffic.  Without mac address filtering, my packet capture results were still very inconsistent (I used -vv to display the packet capture information, but the first packet usually did not appear immediately after I ran the tcpdump command, and the wait time was variable). 
But in one attempt, it took less than a second to fetch 100 packets, something that didn't happen when mac addresses were limited.
After I added the mac address limit, the time to fetch a package was very long and unstable.

XXINCH, 2023-07-22:
@zeroby0 It's still not work.
I have a question, when I use the computer connected to the hot spot to ping the phone ip address at a certain rate, why can't the packet be picked up?
Is it because there is a problem with my mac address filtering? If so, why is it sometimes hard to capture packages when I don't set it.
Regards

zeroby0, 2023-07-24:
No idea why it didn't work. I don't have much experience with other setups, here is what my setup looks like:

1. [TPlink Archer C-20 (AC750)](https://www.amazon.in/TP-Link-Archer-C20-Wireless-Router/dp/B0759QMF85) router
2. Raspberry Pi connected to the router via Ethernet
3. My laptop connected to the router via WiFi

I ssh into the raspberry pi and run an iperf3 server, and an iperf3 client on the laptop to generate traffic and capture it on the pi. I was unable to capture CSI with a different router. Maybe you just need to try out different setups and see what works for you.

XXINCH, 2023-07-26:
@zeroby0 Thanks for your help. I'll try.