seemoo-lab/nexmon_csi #304: Real and Imaginary Parts Calculation

yeqinL, 2023-07-29:
Hi guys,

I am wondering how to calculate real and imaginary parts. I am using Raspberry Pi with bcm43455c0 with channel 6, 2.4GHz and 20 MHz. Here is the 64th subcarrier CSI value. Should 0x0102 be real part and 0x00e8 be imaginary part or 0x0201 be real part and 0xe800 be imaginary part?

[image]

zeroby0, 2023-07-30:
https://github.com/seemoo-lab/nexmon_csi/tree/master/utils/matlab
https://github.com/nexmonster/nexcsi

yjxb1, 2023-10-19:
> 嗨，伙计们，
> 
> 我想知道如何计算实部和虚部。我正在使用带有 bcm43455c0 的树莓派，通道为 6、2.4GHz 和 20 MHz。这是第 64 个子载波 CSI 值。0x0102应该是真实的部分，0x00e8是虚部，还是0x0201实部，0xe800是虚部？此外，它是否有符号位置，这意味着如果更改为十进制，则 65535 应该是 -1。你能告诉我下面CSI值的正确实部和虚部吗？
> 
> [image]
> 
> 干杯！
Hello, have you solved this problem? I also encountered this problem.