seemoo-lab/nexmon_csi #305: tcpdump抓取CSI数据包出错

yjxb1, 2023-08-01:
用的就是该实验室的操作步骤
我用ifconfig指令查找了wlan0MAC地址，也用iwconfig查找到了一个MAC地址，不知道哪一个地址用来替换教程中的地址。
[image]
[image]

然后我将两个MAC地址都试了一遍，信道选的36，结果都不行。伙伴们能解决一下吗
[image]

[image]

sarielkk, 2023-08-17:
Did you solve the problem? I can't make sure which one is my MAC address.
After execute **tcpdump -i wlan0 dst port 5500**, I can't catch any CSI packet. The result is **got 0**.

sangeun-jo, 2023-08-22:
I recommend deactivate or delete wpa_supplicant. 
Mac address filtering is not mandatory option. If you want to use that option,  the mac address must be in channel used in -c option.
Ex) There is the device that address is aa:bb:cc:dd:ee:ff in 36 ch. You can generate base64 parameter like this.
`makecsiparams -N 1 -C 1 -c 36 -m aa:bb:cc:dd:ee:ff`
You can use android app like wifi analyzer.

yjxb1, 2023-08-23:
> 你解决了问题吗？我无法确定哪一个是我的MAC地址。执行 **tcpdump -i wlan0 dst 端口 5500** 后，我无法捕获任何 CSI 数据包。结果**为 0**。

You can use the iwconfig command to view the MAC address。Just like the first picture in my question。

yjxb1, 2023-08-23:
> 我建议停用或删除wpa_supplicant。 Mac 地址过滤不是强制性选项。如果要使用该选项，则 mac 地址必须位于通道中，并且带宽在 -c 选项中使用。例如）有地址为aa：bb：cc：dd：ee：ff的设备，在36通道和20MHz带宽中。您可以像这样生成 base64 参数。 您可以使用像wifi分析仪这样的安卓应用程序来获取特定频道和带宽中的MAC地址。`makecsiparams -N 1 -C 1 -c 36/20 -m aa:bb:cc:dd:ee:ff`

Ok thanks。