seemoo-lab/nexmon_csi #311: How to understand nextmon firmware code

yjxb1, 2023-09-05:
Can everyone help me answer my questions？
1.I am using system version 5.10, so I am using brcmfmac_5.10. What are the various uses of the files in this folder for the nextmon firmware of y-nextmon? How can beginners understand the entire firmware code or become familiar with the entire process.
2.Which file is the code for sending UDP data in?

[image]

zeroby0, 2023-09-05:
I found the free your csi paper very helpful in understanding the code. It's linked in the readme.

yjxb1, 2023-09-05:
> I found the free your csi paper very helpful in understanding the code. It's linked in the readme.

thanks.

jlinktu, 2023-09-05:
`brcmfmac_5.10.y-nexmon` is a slightly modified version of the `brcmfmac` driver from the raspberrypi linux sources version `5.10.y` https://github.com/raspberrypi/linux/tree/rpi-5.10.y
If you want to understand the modifications, try starting with a diff between the nexmon version and https://github.com/raspberrypi/linux/tree/rpi-5.10.y/drivers/net/wireless/broadcom/brcm80211/brcmfmac
But be aware that this is only the driver side running on the host and not part of the firmware.

yjxb1, 2023-09-06:
> `brcmfmac_5.10.y-nexmon` is a slightly modified version of the driver from the raspberrypi linux sources version https://github.com/raspberrypi/linux/tree/rpi-5.10.y If you want to understand the modifications, try starting with a diff between the nexmon version and https://github.com/raspberrypi/linux/tree/rpi-5.10.y/drivers/net/wireless/broadcom/brcm80211/brcmfmac But be aware that this is only the driver side running on the host and not part of the firmware.`brcmfmac``5.10.y`

I am in the csi of the src folder_ I found the xmit() function for sending UDP packets in the extractor file, but I would like to know the content of this function. Where can I find it?
[image]
[image]

jlinktu, 2023-09-06:
This repository does not contain the code behind the function that is assigned to `wl->dev->chained->funcs->xmit`. It is part of the original firmware which is distributed as binary blob. You could analyze the function by disassembling the firmware and/or generating pseudo-code of it with a reversing tool of your choice, e.g. Ghidra.

closed 2023-09-06 by jlinktu