seemoo-lab/nexmon_csi #317: Can't understand how nexmon firmware works

yjxb1, 2023-09-21:
I'm trying to understand the whole working of nexmon.
As shown below, I checked the code in the makecsiparams file under the utils folder and understood the code. Then I want to understand how nexutil works, but I can't find the nexutil code. Does anyone know which folder it is in or where I can find the relevant code?
[image]
Of course I'm doing this just to understand how this firmware extraction filter works, but I don't know where to find the answer I'm looking for.
Can anyone answer my questions? Thank you so much.

zeroby0, 2023-09-21:
Here you go: https://github.com/seemoo-lab/nexmon/tree/master/utilities/nexutil

It's in the main nexmon repo

yjxb1, 2023-09-25:
> Here you go: https://github.com/seemoo-lab/nexmon/tree/master/utilities/nexutil
> 
> It's in the main nexmon repo
Thanks for the answer, I looked at nexutil's readme and it's for Raspberry Pi 3 but I'm using Raspberry Pi 4, does that make a difference?
[image]

zeroby0, 2023-09-25:
> but I'm using Raspberry Pi 4, does that make a difference?

No :)