seemoo-lab/nexmon_csi #318: Enable monitor mode, bcm4355c0 Pi4 5.10.92-v7l

54926, 2023-09-22:
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams# iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
command failed: Operation not supported (-95)

root@raspberrypi:/home/pi# iw dev
phy#0
	Interface mon0
		ifindex 4
		wdev 0x2

		type monitor
		channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 31.00 dBm
	Interface wlan0
		ifindex 3
		wdev 0x1

		type managed
		channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 31.00 dBm

how to solve it, thank you to everyone who provided help.

zeroby0, 2023-09-22:
Restart the Pi and try again? It's the fastest and most reliable 'fix' for me.

Hamzu24, 2024-12-20:
sudo iw dev mon0 del
Fixes it for me