seemoo-lab/nexmon_csi #323: Interpreting the results

SimoneDelPrete, 2023-11-06:
HI, I am trying to use nexmoin csi on a raspberry pi3b. Everything works fine but I have some issues regarding the interpretation of the results.
I am currently doing some tests in the 11 channel (20 MHz), and I am visualizing the results with https://github.com/nexmonster/nexmon_csi/tree/feature/python/utils/python, in particular I simply  run csiexplorer.py and see the results. My questions are: 

1. What are the values in the y-axis? 
2. Why in your opinion I get the high peak in the channel shown in the figure? It is present in all the packets
[image]

Asafe33, 2024-06-05:
could you explain, I'm going through the same situation

SimoneDelPrete, 2024-06-06:
as for the high peak, it's just a matter of selecting the right null and zero subcarriers in the configuration files. I don't exactly remember how, maybe I will look at it, but I remember I found a solution in some discussions.

As for the unit of measure, no answers have been found yet

cledsonsousa, 2024-08-03:
I believe there are no units specified; it is an arbitrary value between 0 and 15000 (amplitude) that needs to be normalized if you intend to convert it to dB or even dBm. The phase values seem correct since they vary from -180 to 180. I have not found the normalization values for either amplitude or phase yet.