seemoo-lab/nexmon_csi #330: Does the csi pattern change when the conditions are all the same? plz share your knewledge.

rack570, 2023-12-18:
Hi, I'm doing research to guess the occupant information with csi. I extracted the csi data when no one is there by writing a repetitive statement code (2 minutes tcpdump, 2 minutes sleep). But the data when no one was there showed different amplitudes. No conditions have changed. How can this happen? I know from the paper that csi can change even when no one is there due to various environmental dynamics. But it's the training data and experimental data on different days, and I only put it 2 minutes apart. If the amplitude value changes every time I try to extract from Raspberry Pi (regardless of the human), can pattern recognition-based csi really be put to practical use?

I've extracted csi and divided it into test and training sets based on its statistical characteristics to evaluate the accuracy and get 96 percent. If the amplitude changes with each csi extraction attempt, I've been producing the wrong result.

jlinktu, 2023-12-18:
The CSI resembles the estimated channel effects between transmitter and receiver at the time a frame is received, including effects of their hardwares. You can't expect it to be stable.
There are numerous publications that deal with separating wanted vs. unwanted effects, you might find something that suits your requirements.
A first step might be to look at relative changes instead of absolute changes of the magnitude over time.

rack570, 2023-12-18:
> The CSI resembles the estimated channel effects between transmitter and receiver at the time a frame is received, including effects of their hardwares. You can't expect it to be stable. There are [...]

Thank you so much for your reply. Are you talking about features similar to values such as std, euc, etc., not these values of amplitude such as max, min, mean? What keywords can I search for to find those papers you are talking about? I am majoring in something that is not very related to this, so I would really appreciate it if you could let me know. There is a lot about machine learning about csi, but I missed the content about wireless signals you are talking about.

jlinktu, 2023-12-18:
You could start with something like "Channel State Information", "post-process" and "sensing" and do a proper literature research. That is something that you have to do on your own.

closed 2023-12-19 by jlinktu