seemoo-lab/nexmon_csi #331: Error: make: *** [Makefile:222: gen/ucode.bin] Error 1

SMuzamilAli, 2023-12-20:
Hi,

I am getting the below error while trying to install the firmware. My kernel version is 6.1.0. When I downgrade it to 5
4 or 4.19, the Raspberry Pi does not boot. Please help me to install it.

make install-firmware

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/muzammil/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PATCHING UCODE src/csi.ucode.bcm43455c0.7_45_189.patch => src/csi.ucode.bcm43455c0.7_45_189.asm
  ASSEMBLING UCODE src/csi.ucode.bcm43455c0.7_45_189.asm => gen/ucode.bin
make: *** [Makefile:222: gen/ucode.bin] Error 1

jlinktu, 2023-12-20:
This might help you: #263 
For the future: Please search existing issues before opening new ones.

SMuzamilAli, 2023-12-21:
> This might help you: #263 For the future: Please search existing issues before opening
@jlinktu Thank you very much for your prompt reply. I have gone through #263, however, when I apply those solutions, my Raspberry Pi does not boot. Can you please let me know do I have to get other Raspberry Pi? what could be the solution.

I tried to downgrade the kernel version.
tried to install python2 but it did not install.

 Thank you.

jlinktu, 2023-12-21:
No need to downgrade.
But you need `python-is-python2`.

jlinktu, 2023-12-21:
Regarding the kernel, from the other issue I see that you are on 6.1.
You can get the driver for your kernel from [here](https://github.com/seemoo-lab/nexmon/tree/master/patches/driver). Put it in the `nexmon_csi` directory and add build instructions [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L112) (by copying from above and exchanging 5.10 with 6.1). Do the same (copying from 5.10 to 6.1) [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L338) and [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L358).

SMuzamilAli, 2023-12-21:
Thanks again for your useful reply. Really appreciate your help. Could you help me with python-is-python2 as well? when I try to install, I get this error:

Package python-is-python2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-is-python3

E: Package 'python-is-python2' has no installation candidate

Really appreciate your support.

jlinktu, 2023-12-21:
The installation script calls a tool that expects python to be of version 2. Please consult the search engine of your choice to figure out how to get the environment in that you are executing the Makefile to use python version 2. There are plenty of resources out there about this.

jlinktu, 2023-12-21:
The tool we are talking about uses a shebang [here](https://github.com/seemoo-lab/nexmon/blob/29d990bdaea869cb32f5eaa2dbb7efacdcd405aa/buildtools/b43-v3/debug/b43-beautifier#L1). So if you manage to install python2 on your system you can also edit this to use version 2 directly.

zeroby0, 2023-12-22:
> Regarding the kernel, from the other issue I see that you are on 6.1. You can get the driver for your kernel from [here](https://github.com/seemoo-lab/nexmon/tree/master/patches/driver). Put it in [...]

Wow I can run nexmon_csi on kernel 6 if I do that? Time to start compiling new binaries then! :D

SMuzamilAli, 2023-12-23:
> Regarding the kernel, from the other issue I see that you are on 6.1. You can get the driver for your kernel from [here](https://github.com/seemoo-lab/nexmon/tree/master/patches/driver). Put it in [...]

@jlinktu I am sorry for bothering you again and again. I am new in this field and getting issues while installing nexmon. I have managed to get python 2 however, I am unable to find the location of the Makefiles you mentioned here, where I have to replace the 5.10 with 6.1.

Kindly help me to locate those folders where the files are located.

Thank you very much for your time and support.

SMuzamilAli, 2023-12-23:
@jlinktu Sorry for bothering you again. I have made the changes as per your suggestion and started from the beginning. However, source setup_env.sh, when I run make, it give me following error. Could you please help me with that? I have installed python2 already and made changes as per your suggestions.

I get this error:  

EXTRACTING UCODE
ERR: ram file empty or unavailable.
make[3]: *** [Makefile:17: ucode.bin] Error 1
make[3]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0/7_95_49_2271bb6'
make[2]: *** [Makefile:8: 7_95_49_2271bb6/] Error 2
make[2]: Leaving directory '/home/pi/nexmon/firmwares/bcm43439a0'
make[1]: *** [Makefile:7: bcm43439a0/] Error 2
make[1]: Leaving directory '/home/pi/nexmon/firmwares'
make: *** [Makefile:5: firmwares] Error 2

I have already gone through https://github.com/seemoo-lab/nexmon/issues/578 and https://github.com/seemoo-lab/nexmon_csi/issues/32 but no help.

Kindly help me to get out of this situation.

Thank you very much for your time and support.

jlinktu, 2023-12-23:
Maybe your system is missing `xxd`.
If that is the case please install it and retry.

If the above doesn't help, please do the following and report the full output:
```
cd firmwares/bcm43439a0/7_95_49_2271bb6
make
```

SMuzamilAli, 2023-12-23:
Thank you so much for your prompt reply. I have installed xxd but getting same error. 

after running cd firmwares/bcm43439a0/7_95_49_2271bb6
make 

getting this error:

  EXTRACTING UCODE
ERR: ram file empty or unavailable.
make: *** [Makefile:17: ucode.bin] Error 1

jlinktu, 2023-12-23:
Run `make clean` first and then `make`.

SMuzamilAli, 2023-12-24:
@jlinktu Thank you so much. It has worked but at the end I have got this warning. Could you please let me know what is this?

  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
Makefile:116: Warning: Driver can not be compiled on this platform, execute the make command on a raspberry pi
Makefile:340: Warning: Cannot install firmware on this arch., bcm43430-sdio.bin needs to be copied manually into /lib/firmware/brcm/ on your RPI3

You are a great support. Really appreciate it.

zeroby0, 2023-12-24:
The latest raspbian uses a 64-bit kernel and a 32-bit userland.  Nexmon_CSI is not supported for 64-bit kernels (afaik), so you'll need to switch over to a 32-bit kernel. Add `arm_64bit=0` at the end of the `/boot/config.txt` file and reboot and start over.

I did that 2 days ago and got working binaries, but they worked only once and I got a nexutil error every time after reboot. I'll take another look at it soon and upload them to nexmon_csi_bin after testing.

liuchengzhi314159, 2024-09-25:
>

> Maybe your system is missing `xxd`. If that is the case please install it and retry.
> 
> If the above doesn't help, please do the following and report the full output:
> 
> ```
> cd firmwares/bcm43439a0/7_95_49_2271bb6
> make
> ```
Excuse me, I may have a similar problem with the questioner. A new issue has been raised.
I need to install nexmon_csi, (aarch64 architecture, linux kernel 6.1) on a raspberry pi 4b. If I understand your answer above correctly, I need to copy the nexmon_csi folder to the /patches/bcm4.455c0/7_45_189 directory, and then copy the 6.1 driver folder from the drivers folder in the /patches/bcm4.455c0/7_45_189 directory to the nexmon_csi folder, and then change all 5.10 to 6.1 in the Makeflie file.
At the same time I have a question, because the soc architecture in the makefile in this folder is armv7, do I just add an if branch, aarch64 architecture case? And then what do I do to fix the python2 problem? Modify the makefile?

liuchengzhi314159, 2024-09-25:
> > Regarding the kernel, from the other issue I see that you are on 6.1. You can get the driver for your kernel from [here](https://github.com/seemoo-lab/nexmon/tree/master/patches/driver). Put it in the `nexmon_csi` directory and add build instructions [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L112) (by copying from above and exchanging 5.10 with 6.1). Do the same (copying from 5.10 to 6.1) [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L338) and [here](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L358).
> 
> Wow I can run nexmon_csi on kernel 6 if I do that? Time to start compiling new binaries then! :D
> 
> Update: `uname -m` reports aarch64 on the latest raspbian while `dpkg --print-architecture` reports armhf. https://forums.raspberrypi.com/viewtopic.php?p=2104686#p2104686

Excuse me, I may have a similar problem with the questioner. A new issue has been raised.
I need to install nexmon_csi, (aarch64 architecture, linux kernel 6.1) on a raspberry pi 4b. If I understand your answer above correctly, I need to copy the nexmon_csi folder to the /patches/bcm4.455c0/7_45_189 directory, and then copy the 6.1 driver folder from the drivers folder in the /patches/bcm4.455c0/7_45_189 directory to the nexmon_csi folder, and then change all 5.10 to 6.1 in the Makeflie file.
At the same time I have a question, because the soc architecture in the makefile in this folder is armv7, do I just add an if branch, aarch64 architecture case? And then what do I do to fix the python2 problem? Modify the makefile?

closed 2025-12-01 by jlinktu