seemoo-lab/nexmon_csi #332: Question about phy_reg of BCM4366c0

Mojuscran, 2023-12-26:
Hi,
My recent work involves hardware register addresses, where can I find the physical register address table for the bcm4366c0, I can't find it on the Internet. I would be grateful if anyone could help me.

jlinktu, 2023-12-26:
Hi,
what is your goal?

Mojuscran, 2023-12-26:
I want to modify patch to get the value of the AGC register, could you give me some hints?

jlinktu, 2023-12-26:
Someone wrote this for the `bcm43455c0`: https://github.com/T3rO/nexmon_csi_gain
A modification that reads out lots of gain related stuff including an AGC register, which is what you are probably looking for. It is likely that the register address is the same for the other chips.

Mojuscran, 2023-12-26:
The author of [https://github.com/T3rO/nexmon_csi_gain](url) mentioned that he looked up the register address through the `wlc_phyreg_ac.h` in [https://github.com/MerlinRdev/86u-merlin/blob/master/release/src-rt-5.02hnd/bcmdrivers/broadcom/net/wl/impl51/4365/src/wl/phy/wlc_phyreg_ac.h](url), But the register addresses `0x872-0x874` involved in the modification of [https://github.com/seemoo-lab/nexmon_csi/issues/126#issuecomment-704981416](url) represent different meanings in `wlc_phyreg_ac.h`, it seems that it does not represent RXPWR. I would like to know how you know the addresses of certain registers.

Deltaaaaaaaaaa, 2023-12-26:
Hi, @jlinktu,
Thank you so much for your guidance!

I am also working on this problem. I am exploring an alternative approach to ensure alignment between the AGC value and the CSI. Your work on [issue #126](https://github.com/seemoo-lab/nexmon_csi/issues/126#issuecomment-704981416), particularly the addition of the rxpwr value at the beginning of the CSI frames, has been a great inspiration to me. Currently, I'm trying to identify the register address for the AGC gain codes.

In your patch, I noticed the use of addresses 0x872 to 0x874 to extract the rxpwr value. However, according to the "wlc_phyreg_ac.h" and "d11.h" files, these addresses seem to be allocated for padding in tx registers, which is quite puzzling to me. It raises questions about how to correctly identify the register addresses using these header files. Additionally, I'm not entirely certain that I'm referencing the correct versions of these files.

Could you please offer more detailed instructions or clarification regarding the register addresses? I would greatly appreciate any further insight you could provide.

jlinktu, 2023-12-26:
The values `0x872`, `0x873`, and `0x874` in the patch that I once uploaded [here](https://github.com/seemoo-lab/nexmon_csi/issues/126#issuecomment-704981416) are not `PHY`-registers. Those are locations of the `SHM` (shared memory). Which at the point where they are used in the patch hold values that are copied from some `IHRs` (internal hardware registers). These specific `IHRs` hold something called `PHY receive status` (`PhyRxStatus`, `PHYRS`), which i.a. hold receive power values. Hence, all we do in the patch there is to read the receive power values from these headers. The `IHRs`, `SHM` locations and the bit mappings can be found from `d11.h`, e.g. [here](https://github.com/RMerl/asuswrt-merlin/blob/263449f32bf292fb6bc5a08cd645e61a7fb10485/release/src-rt-7.14.114.x/src/include/d11.h#L2550), and from analyzing the `ucode`.

jlinktu, 2023-12-27:
For the `bcm4366c0` there is a function that reads out gain related infos. It is called `wlc_phy_get_rxgain_acphy` and is located at `0x25C156` for the fw version 10.10.122.20. It uses all the addresses that you are looking for.
But instead of re-implementing everything like it is done in the `nexmon_csi_gain` repo, you could also just call the function. E.g. like this:
```C
struct rxgain {
    uint8 lna1;
    uint8 lna2;
    uint8 mix;
    uint8 lpf0;
    uint8 lpf1;
    uint8 dvga;
} __attribute__((packed));

#define N_GAIN_TYPES 8
#define N_ANTENNAS 4

uint8 force_gain_types[N_GAIN_TYPES] = {1, 2, 3, 4, 6, 7, 8, 9};

int16 tot_gain[N_GAIN_TYPES][N_ANTENNAS];
struct rxgain rxgain[N_GAIN_TYPES][N_ANTENNAS];

int i;
for (i = 0; i < N_GAIN_TYPES; i++) {
    wlc_phy_get_rxgain_acphy(wlc->hw->band->pi, rxgain[i], tot_gain[i], force_gain_types[i]);
}
```
_I already added the function address to the nexmon repo._

Mojuscran, 2023-12-27:
@jlinktu Thank you for providing the solution. I greatly appreciate your work.

closed 2023-12-27 by jlinktu

Mojuscran, 2024-01-02:
@jlinktu
I'm sorry to bother you again. Does each core of bcm4366c0 have an `AGC` module? I saw the definition `int16 agc_gain[PHY_CORE_NUM_4];` in `wlc_phy_ac.c` file. But the function `wlc_phy_agc_acphy`, which is used to obtain the `AGC` values, is not defined. I tried accessing it by searching for the register addresses, but I couldn't find the addresses😢. I would like to know how to retrieve these four AGC values. Could you provide me with their register addresses or how to find them? If you could spare some time to help me amidst your busy schedule, I would be extremely grateful.