seemoo-lab/nexmon_csi #338: Does anyone know about the range or coverage of Wi-Fi wireless transmission?

rack570, 2024-02-02:
Hi, I'm a nexmon csi extractor user. I want to use csi to estimate whether each room is occupied in two adjacent rooms. If Tx is in the middle and there are people in Room 2 receiving csi from Rx1 and Rx2, the csi variation will be detected severely. On the contrary, however, if there are people in Room 1, there is some variation in Rx2, but it is not found very much. Why is that? Do you have any literature on the range form of radio waves? Is it not appropriate to put Tx between Rxs in a straight line? I would appreciate it if you could share your knowledge on the coverage of radio waves.

Note: Room1 is a private room and Room2 is a common area (ex. kitchen)

rack570, 2024-02-06:
@jlinktu  Oh, thanks for replying.
By the way, is Nexmon csi extractor still an unstable program?
I always run the same command in the same situation, sometimes I continuously extract csi for more than 24 hours, but sometimes it stops during packet extraction. I don't think it's the Raspberry Pi device's own problem. Do u know anything about this?

rack570, 2024-02-07:
@jlinktu  Hi, this is my dmesg output. I don't see any crashes, just went into promiscuous mode and didn't come out.
For your information, I've written a repeat script. I'll show you the kernel input window that was suspended during the script and tcpdump. Let me know if you need anything else. For your information, if I'm suspended like this, no matter what I do, such as Chanspec or reset the MAC address, csi packet will be captured. It's only possible by reboot.

device = raspberry pi 4B
kernel version = 5.10.92
Raspbian = Raspbian Buster Lite 2022-01-28
Chip = BCM43455c0

[image]

[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 34360 entries in 101 pages
[    0.000000] ftrace: allocated 101 pages with 4 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=1
[    0.000008] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000035] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000095] bcm2835: system timer (irq = 26)
[    0.000757] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000779] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000802] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000819] Switching to timer-based delay loop, resolution 18ns
[    0.001092] Console: colour dummy device 80x30
[    0.001848] printk: console [tty1] enabled
[    0.001914] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001968] pid_max: default: 32768 minimum: 301
[    0.002146] LSM: Security Framework initializing
[    0.002348] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002394] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003871] cgroup: Disabling memory control group subsystem
[    0.004124] CPU: Testing write buffer coherency: ok
[    0.004602] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005827] Setting up static identity map for 0x200000 - 0x20003c
[    0.006041] rcu: Hierarchical SRCU implementation.
[    0.006988] smp: Bringing up secondary CPUs ...
[    0.008212] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.009581] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.010982] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.011143] smp: Brought up 1 node, 4 CPUs
[    0.011190] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.011220] CPU: All CPU(s) started in HYP mode.
[    0.011246] CPU: Virtualization extensions available.
[    0.012098] devtmpfs: initialized
[    0.026493] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.026731] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.026784] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.035026] pinctrl core: initialized pinctrl subsystem
[    0.036127] NET: Registered protocol family 16
[    0.040101] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.040976] audit: initializing netlink subsys (disabled)
[    0.041252] audit: type=2000 audit(0.040:1): state=initialized audit_enabled=0 res=1
[    0.041840] thermal_sys: Registered thermal governor 'step_wise'
[    0.042509] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.042563] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.042984] Serial: AMBA PL011 UART driver
[    0.087036] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.100923] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-01-20T13:56:48, variant start
[    0.110936] raspberrypi-firmware soc:firmware: Firmware hash is bd88f66f8952d34e4e0613a85c7a6d3da49e13e2
[    0.151963] Kprobes globally optimized
[    0.157538] bcm2835-dma fe007000.dma: DMA legacy API manager, dmachans=0x1
[    0.162599] vgaarb: loaded
[    0.163086] SCSI subsystem initialized
[    0.163341] usbcore: registered new interface driver usbfs
[    0.163417] usbcore: registered new interface driver hub
[... 350 lines cut ...]
[  513.715934] device wlan0 entered promiscuous mode
[  533.442951] device wlan0 left promiscuous mode
[  573.675968] device wlan0 entered promiscuous mode
[  593.443722] device wlan0 left promiscuous mode
[  633.693354] device wlan0 entered promiscuous mode
[  653.443688] device wlan0 left promiscuous mode
[  693.708068] device wlan0 entered promiscuous mode
[  713.445849] device wlan0 left promiscuous mode
[  753.718999] device wlan0 entered promiscuous mode
[  773.445412] device wlan0 left promiscuous mode
[  813.690579] device wlan0 entered promiscuous mode
[  833.445884] device wlan0 left promiscuous mode
[  873.683875] device wlan0 entered promiscuous mode
[  893.446219] device wlan0 left promiscuous mode
[... cut ...]
[  953.446321] device wlan0 left promiscuous mode
[  993.663987] device wlan0 entered promiscuous mode
[ 1013.457120] device wlan0 left promiscuous mode
[ 1053.654373] device wlan0 entered promiscuous mode
[ 1073.447614] device wlan0 left promiscuous mode
[ 1113.711574] device wlan0 entered promiscuous mode
[ 1133.448362] device wlan0 left promiscuous mode
[ 1173.675448] device wlan0 entered promiscuous mode
[ 1193.450528] device wlan0 left promiscuous mode
[ 1233.684057] device wlan0 entered promiscuous mode

jlinktu, 2024-02-08:
Unfortunately, this doesn't help in identifying the issues.

You can try to make sure not being in sleep mode with: `sudo iw dev wlan0 set power_save off`, but it can be that this doesn't make any difference as the chip should not sleep anyways.

When such a hang-up occurs again, please run the following and post the output:
```
sudo -u root cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | awk '/wiphy/ {printf "phy" $2}')/forensics; echo
```

rack570, 2024-02-11:
@jlinktu 
Hi, thank you for continuing to reply. I was in the middle of trouble because I was constantly getting errors. I didn't expect to have trouble extracting csi before I even solved the problem with csi data processing. XD

Just like you said
"sudo iw dev wlan0 set power_save off"
I wrote the script by adding this command from the original script. But the error is still continuing. I'll give you a general command that might help you. I really hope this is resolved.

Note: I don't think it's a "sleep" issue. Instead of a script that sets this extraction cycle, I've had csi continuously extracted and observed it. This time, I was getting about 9300 packets, but stopped along the way. My sampling rates are 25Hz, and it's a beacon frame, not an artificial ICMP ping. Is the beacon signal an issue?

pi@raspberrypi:~ $ sudo su
root@raspberrypi:/home/pi# chmod +x nexmon_v3_5G_2.sh
root@raspberrypi:/home/pi# ./nexmon_v3_5G_2.sh
hihi zz nexmon extractor gogo
Enter the value of k : 144
Enter capture duration (second) : 20
Enter sleep duration over 60s (second) : 280
Enter the time (ex.hhmmss) : 193500
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
457 packets captured
482 packets received by filter
0 packets dropped by kernel
Saved file number zz : 1 of 144
Next desired time: 194000
Sleeping for 252 seconds... & Waiting for 28 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
460 packets captured
485 packets received by filter
0 packets dropped by kernel
Saved file number zz : 2 of 144
Next desired time: 194500
Sleeping for 252 seconds... & Waiting for 28 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
456 packets captured
481 packets received by filter
0 packets dropped by kernel
Saved file number zz : 3 of 144
Next desired time: 195000
Sleeping for 252 seconds... & Waiting for 28 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
481 packets captured
506 packets received by filter
0 packets dropped by kernel
Saved file number zz : 4 of 144
Next desired time: 195500
Sleeping for 252 seconds... & Waiting for 28 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
458 packets captured
482 packets received by filter
0 packets dropped by kernel
Saved file number zz : 5 of 144
Next desired time: 200000
Sleeping for 252 seconds... & Waiting for 28 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
^Zt 0                                                   --------------------------------------------------------> Stop command due to error
[1]+  Stopped                 ./nexmon_v3_5G_2.sh
root@raspberrypi:/home/pi# sudo -u root cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | awk '/wiphy/ {printf "phy" $2}')/forensics; echo
hndarm_armr addr: 0x18002000, cr4_idx: 0
000000.001
RTE (SDIO-CDC) 7.45.189 (r714228 CY) on BCM4345 r6 @ 37.4/160.2/160.2MHz
000000.001 nvram_init: called again without calling nvram_exit()
000000.002 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.002 nvram_init: called again without calling nvram_exit()
000000.002 Decompressing ucode at 002207a4 (len: 28657) at 0025a4e0
000000.146 Decompression res:0
000000.147 reclaim section 0: Returned 41892 bytes to the heap
000000.149 wlc_bmac_info_init: host_enab 0
000000.149 Overriding vendor id = 0x14e4
000000.149 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.199 wl0: wlc_stf_txcore_shmem_write: No clock
000000.200 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.205 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.206 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.207 TCAM: 256 used: 237 exceed:0
000000.208 reclaim section 1: Returned 118076 bytes to the heap
000000.208 reclaim section 4: Returned 44 bytes to the heap
000000.208 sdpcmd_dpc: Enable
000000.246 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.249 wl0: unable to find iovar "rsdb_mode"
000000.249 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.250 wl0: unable to find iovar "tdls_enable"
000000.250 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000036.327 wl0: unable to find iovar "toe_ol"
000036.327 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000036.327 wl0: wl_open
000036.338 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000036.508 wl0: unable to find iovar "toe_ol"
000036.508 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
root@raspberrypi:/home/pi#

jlinktu, 2024-02-12:
Nothing in there that indicated a crash or misbehavior of the firmware.

Maybe interference with a different program that tries to access the wifi chip, such as `wpa_supplicant` or a network manager? Is the channel of your choice still set when the capture stops? (you can check with `nexutil -k`)

Can you rule out that your transmitter is misbehaving?

rack570, 2024-02-13:
@jlinktu 
The extraction just stopped, so I checked the` nexutil -k` you mentioned, and it's the same channel 44 I set initially. And I have 3 transmitters in total, just in case, I've already checked them and these errors are constantly occurring for all transmitters. `wpa_supplicant` is not running. If there is no problem with the firmware, is it because of the instability of the raspberry pie due to the device load? I purposely added the sleep command to stop the load. However, when I asked ( nexmonster/nexmon_csi ), they answered that they had tried it for up to 16 hours. hmm,. I'm having a lot of worries.

Could you please check this out? This is the result of running `sudo -u root cat /sys/ kernel/debug/ieeee80211/$ (iw dev wlan0 info | awk '/wiphy/{printf "py" $2})/forensics; echo`.
There's something different from the output I gave you before.
"wl0: dma_rx: bad frame length"

Saved file number zz : 43 of 100
Next desired time: 115400
Sleeping for 36 seconds... & Waiting for 4 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
Maximum file limit reached: 1
1404 packets captured
1480 packets received by filter
0 packets dropped by kernel
Saved file number zz : 44 of 100
Next desired time: 115500
Sleeping for 36 seconds... & Waiting for 4 seconds...
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
^Zt 0     ---------------------------------------------------> csi packets capture stopped
[1]+  Stopped                 ./nexmon_v3.sh
root@raspberrypi:/home/pi# nexutil -k
chanspec: 0xd02c, 44
root@raspberrypi:/home/pi# sudo -u root cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | awk '/wiphy/ {printf "phy" $2}')/forensics; echo
lled again without calling nvram_exit()
000000.002 sdpcmdcdc0: Broadcom SDPCMD CDC driver
000000.002 nvram_init: called again without calling nvram_exit()
000000.002 Decompressing ucode at 002207a4 (len: 28657) at 0025a4e0
000000.146 Decompression res:0
000000.147 reclaim section 0: Returned 41892 bytes to the heap
000000.149 wlc_bmac_info_init: host_enab 0
000000.149 Overriding vendor id = 0x14e4
000000.149 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.199 wl0: wlc_stf_txcore_shmem_write: No clock
000000.200 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.205 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.206 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.207 TCAM: 256 used: 237 exceed:0
000000.208 reclaim section 1: Returned 118076 bytes to the heap
000000.208 reclaim section 4: Returned 44 bytes to the heap
000000.208 sdpcmd_dpc: Enable
000000.224 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.227 wl0: unable to find iovar "rsdb_mode"
000000.227 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.228 wl0: unable to find iovar "tdls_enable"
000000.228 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000103.756 wl0: unable to find iovar "toe_ol"
000103.756 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000103.756 wl0: wl_open
000103.768 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000103.928 wl0: unable to find iovar "toe_ol"
000103.928 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000747.342 wl0: dma_rx: bad frame length (1651)
000924.803 wl0: dma_rx: bad frame length (1698)
002392.853 wl0: dma_rx: bad frame length (1682)
002453.845 wl0: dma_rx: bad frame length (1682)

jlinktu, 2024-02-13:
Seems not to be related to your problem.

rack570, 2024-03-14:
Hello @jlinktu 

I think I might have found something regarding this phenomenon. When extracting data at 80MHz, I can receive CSI continuously without interruptions, but when I set it to 20MHz or 40MHz, it stops partway through. I have insisted on using 20MHz because I thought that 256 subcarriers were too many. Could there be a software bug when extracting CSI at 20MHz or 40MHz?