seemoo-lab/nexmon_csi #339: Issue about frame injection with Nexus 5 using Jammer APP

pygj1994, 2024-03-11:
Hello, thank you for your good work, they have been very helpful to me. Now, when I execute /jffs/tcpdump -i eth6 dst port 5500, I can already collect frames with RT-AC86U. Now, I want to use Nexus 5 to inject frames, so I installed the Jammer APP on Nexus 5, selected the transmitter mode, where the destination port is set to 3939, modulation is set to 802.11ac, MCS is 0, bandwidth is 80MHz, channel 36 (5180MHz) but after I click on transmit, the RT-AC86U does not receive any packets. The commands I executed are /jffs/mcp -c 36/80 -C 1 -N 1; /jffs/nexutil -Ieth6 -s500 -b -l34 -vKuABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==; /usr/sbin/wl -i eth6 monitor 1; /jffs/tcpdump -i eth6 dst port 3939. I wonder if my configuration is correct? I hope to receive your reply, thank you very much!

jlinktu, 2024-03-11:
I assume from your description that you want to extract CSI on the Asus RT-AC86U that are created when frames that are injected to the medium from an Google Nexus 5 are overheard. As the [README](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#analyzing-the-csi) tells, extracted CSI are transported from the WiFi chip to the host via UDP frames on port 5500. Thus, you want to listen on port 5500 instead of 3939.

pygj1994, 2024-03-12:
> I assume from your description that you want to extract CSI on the Asus RT-AC86U that are created when frames that are injected to the medium from an Google Nexus 5 are overheard. As the [README](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#analyzing-the-csi) tells, extracted CSI are transported from the WiFi chip to the host via UDP frames on port 5500. Thus, you want to listen on port 5500 instead of 3939. The port that you select in the Jammer App is not related to the port where the CSI UDPs arrive.

Thank you very much for your answer. I have now adjusted the port number to 5500 and used the Jammer APP to inject frames. On the RT-AC86U side, it shows: 

admin@RT-AC86U-F0A8:/tmp/home/root# /jffs/tcpdump -i eth6 dst port 5500
tcpdump: WARNING: eth6: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth6, link-type EN10MB (Ethernet), capture size 65535 bytes
13:23:37.779548 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
13:23:37.790734 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
13:23:37.793527 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042

However, it seems that these are not frames emitted from the Jammer app because after I changed the fps in the Jammer app to 1, the receiving rate on the RT-AC86U side does not change. How can I select the frames sent out by the Jammer app?

Additionally, my current setup is as follows: the RT-AC86U is connected to my laptop via an Ethernet cable, and the RT-AC86U is also connected to an AP via another Ethernet cable. The Nexus 5 with the Jammer app installed is set to transmitter mode and continuously emits packets. Apart from this, there are no other devices involved. Is this configuration reasonable? Thank you once again!

pygj1994, 2024-03-12:
> > How can I select the frames sent out by the Jammer app?
> 
> The format and partly content of the frames that are sent out by the Google Nexus 5 using the Jammer App are defined [here](https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/8eb32a8b96bf42b6e2c4a0ffa4db77a8cf04ebc8/src/ioctl_5xx.c#L178). From this, one can see that the type/subtype field is `0x88` and the source address is 'JAMMER' (`4a:41:4d:4d:45:52`). You can configure the CSI extractor on the Asus RT-AC86U to only consider frames that are of a specific type/subtype and have a specific source address. This could look like the following:
> 
> ```
> /jffs/mcp -c 36/80 -C 1 -N 1 -b 0x88 -m 4a:41:4d:4d:45:52
> KuABEQGIAQBKQU1NRVIAAAAAAAAAAAAAAAAAAAAAAAAAAA==
> ```
> 
> > Additionally, my current setup is as follows: the RT-AC86U is connected to my laptop via an Ethernet cable, and the RT-AC86U is also connected to an AP via another Ethernet cable. The Nexus 5 with the Jammer app installed is set to transmitter mode and continuously emits packets. Apart from this, there are no other devices involved. Is this configuration reasonable? Thank you once again!
> 
> As far as I can tell from the description, yes, seems reasonable. At least I don't see why it shouldn't be reasonable.

Thank you very much for your response. I have tried the command you mentioned: /jffs/mcp -c 36/80 -C 1 -N 1 -b 0x88 -m 4a:41:4d:4d:45:52, and I set  Jammer app's transmitter mode to  channel 36 and bandwidth to 80. However, when executing /jffs/tcpdump -i eth6 dst port 5500, the CSI extractor on the RT-AC86U side does not capture any packets. I'm not sure if this is because the frames transmitted by the Jammer app are not stable enough. This is because when I use another Nexus 5 installed with the Jammer app as the receiver, sometimes the receiver can receive packets, and it will receive packets even if the frequency does not match with the transmitter;  however, sometimes the receiver cannot receive any packets at all. For example, in the picture, no matter how the channel and bandwidth are set on the receiver side, it can always receive packets on port 3939 (Transmitter keeps as channel 36 and bandwidth 80). The bandwidth displayed in the receiver's table is always 20MHz, so I'm not sure if the channel and bandwidth settings on the RT-ACU side are inconsistent with the actual transmission, or if there is some other reason, which is very confusing to me. Thank you once again! 

[image]

jlinktu, 2024-03-13:
You can check if the channels are correctly set on the Google Nexus 5 and the Asus RT-AC86U.
On the Google Nexus 5 (make sure `nexutil` is installed):
```
$ nexutil -k
```
On the Asus RT-AC86U:
```
$ wl -i eth6 chanspec
```
Both should give you `36/80`.

To check if the Asus router can receive the frames that are sent out by the Nexus 5 you can configure it as a monitor device and check for the expected frames as described in the "Setup `mon`" part [here](https://github.com/seemoo-lab/nexmon_tx_task/issues/1#issuecomment-1438403189).

closed 2024-03-13 by jlinktu