seemoo-lab/nexmon_csi #341: About the installation of bcm4366c0

maple-42, 2024-04-16:
I am now attempting to install bcm4366c0 though,
7.Enter the created subdirectory nexmon_csi and run make install-firmware REMOTEADDR=<address of your rt-ac86u> to compile our firmware patch and install it on your RT-AC86U router.

I am having trouble with the error below when executing here.
bash: syntax error near unexpected token `newline'

How would you install it on the RT-AC86U router, and if there is a connection method, etc., I would appreciate it if you could tell me.

Thank you in advance.

maple-42, 2024-04-26:
I was able to resolve the error I wrote about earlier, but now I get this error.
I hope you can help me with this, as reading #263 did not solve the problem for me.

 COLLECTING STATISTICS read /home/user/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  ASSEMBLING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.asm => gen/ucode.bin
make: *** [Makefile:222: gen/ucode.bin] Error 1

@zeroby0

jlinktu, 2024-04-26:
For others it might be beneficial if you document the solution to your previous problem here.

Regarding your current error, please post the output of `nexmon_csi/log/ass.log`.

maple-42, 2024-04-26:
Thank you very much.

I ran nexmon_csi/log/ass.log and got the following output.

bash: nexmon_csi/log/ass.log: No such file or directory

jlinktu, 2024-04-26:
Ok, we talked past each other here.
`ass.log` is a log file that gets produced during the step that is failing for you. You can find it your `nexmon_csi` patch directory in the `log` sub-directory.
You can get the file's content for example like this (assuming you have NEXMON_ROOT in your environment / have executed `source setup_env.sh` in the same environment): 
```bash
cat "$NEXMON_ROOT/patches/bcm4366c0/10_10_122_20/nexmon_csi/log/ass.log"
```

maple-42, 2024-04-26:
Sorry, I am a beginner and do not understand well.

When I opened the ass.log, this is what I found.

Parser ERROR (file "src/csi.ucode.bcm4366c0.10_10_122_20.asm", line 125, col 9):
 xje [SHM(0x1724)], 0, skipdeaf+
        ^
syntax error

Thank you for your reply.

maple-42, 2024-04-29:
I will try again from the beginning. Thank you very much.

maple-42, 2024-04-29:
I was retrying and another error occurred.

COLLECTING STATISTICS read /home/user/nexmon/STATISTICS.md for more information
  COMPILING src/console.c => obj/console.o (details: log/compiler.log)
  COMPILING src/csi_extractor.c => obj/csi_extractor.o (details: log/compiler.log)
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
  COMPILING src/patch.c => obj/patch.o (details: log/compiler.log)
  COMPILING src/regulations.c => obj/regulations.o (details: log/compiler.log)
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  DISASSEMBLING UCODE /home/user/nexmon/firmwares/bcm4366c0/10_10_122_20/nonmuucode.bin => gen/ucode.asm
/usr/bin/env: ‘python’: No such file or directory
make: *** [Makefile:213: gen/ucode.asm] Error 127

jlinktu, 2024-04-29:
Install `python2` and change [this line ](https://github.com/seemoo-lab/nexmon/blob/aa9043e8b301e9392afc837b33c9b40f6c6aedfe/buildtools/b43-v2/debug/b43-beautifier#L1) to `#!/usr/bin/env python2`.

maple-42, 2024-05-06:
Thank you, I installed python2 and tried again, but I got the following error.

  COLLECTING STATISTICS read /home/user/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PATCHING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.patch => src/csi.ucode.bcm4366c0.10_10_122_20.asm
  ASSEMBLING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.asm => gen/ucode.bin
make: *** [Makefile:222: gen/ucode.bin] Error 1

maple-42, 2024-05-06:
I modified it in my own way and it changed to this error.

ssh: connect to host ***.***.**.* port **: Connection refused
lost connection
make: *** [Makefile:365: install-firmware] Error 1

maple-42, 2024-05-06:
I also encountered this error.

sh: scp: not found
lost connection
make: *** [Makefile:365: install-firmware] Error 1

lionlanlan, 2024-06-08:
> I also encountered this error.
> 
> sh: scp: not found lost connection make: *** [Makefile:365: install-firmware] Error 1

> I also encountered this error.
> 
> sh: scp: not found lost connection make: *** [Makefile:365: install-firmware] Error 1

> Thank you, I installed python2 and tried again, but I got the following error.
>
> COLLECTING STATISTICS read /home/user/nexmon/STATISTICS.md for more information COMPILING src/version.c => [...]

> Thank you, I installed python2 and tried again, but I got the following error.
>
> COLLECTING STATISTICS read /home/user/nexmon/STATISTICS.md for more information COMPILING src/version.c => [...]

hi，I also met this problem and did you have slove it？Thx

jlinktu, 2024-06-08:
`scp` is no longer available on the Asus AC86U with recent firmware images. You can either use an older firmware image that comes with `scp` or you can exchange the following line:
https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L365
with an alternative to copy the file over to the device, e.g.:
```
(Q)ssh admin@$$REMOTEADDR '/usr/sbin/socat -u TCP-LISTEN:65535,reuseaddr OPEN:/jffs/dhd.ko,creat'& sleep 1; socat -u FILE:dhd.ko TCP:$$REMOTEADDR:65535
```
Note that this requires you to have `socat` installed on your machine.

lionlanlan, 2024-06-11:
> `scp` is no longer available on the Asus AC86U with recent firmware images. You can either use an older firmware image that comes with `scp` or you can exchange the following line:
> 
> https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L365
> 
> with an alternative to copy the file over to the device, e.g.:
> 
> ```
> (Q)ssh admin@$$REMOTEADDR '/usr/sbin/socat -u TCP-LISTEN:65535,reuseaddr OPEN:/jffs/dhd.ko,creat'& sleep 1; socat -u FILE:dhd.ko TCP:$$REMOTEADDR:65535
> ```in 
> 
> Note that this requires you to have `socat` installed on your machine.

Thank you very much and I have done the getting started succesfully but when I went to the run 'nexutil'  in the terminal of AC86U got this:
admin@RT-AC86U-5360:/jffs# nexutil
/jffs/nexutil: line 1: ELF: not found
/jffs/nexutil: line 2: syntax error: unexpected word (expecting ")")

can you help me or give me some advice to solve this problem? thx!!

jlinktu, 2024-06-11:
That indicates that you compiled `nexutil` for the wrong architecture. Make sure to build it by following steps 8, 9, and 10 of the README's section bcm4366c0: https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#bcm4366c0

lionlanlan, 2024-06-17:
Thank you. I have solved it by your advice.

closed 2024-06-17 by jlinktu

maple-42, 2024-07-02:
I rewrote what was stated and ran it, but the error occurred again.
 
 COPYING TO ROUTER dhd.ko => /jffs/dhd.ko
(Q)ssh admin@$REMOTEADDR '/usr/sbin/socat -u TCP-LISTEN:65535,reuseaddr OPEN:/jffs/dhd.ko,creat'& sleep 1; socat -u FILE:dhd.ko TCP:$REMOTEADDR:65535
/bin/sh: 1: Syntax error: word unexpected
make: *** [Makefile:365: install-firmware] エラー 2

jlinktu, 2024-07-02:
There was a `$` missing at the beginning of the command. Please don't blindly copy&paste, always recheck commands, this one could have been easily spotted.

maple-42, 2024-07-02:
Sorry, thank you very much.
 I apologize for the trouble.

maple-42, 2024-07-04:
Sorry, I have been trying to find out how to fix the code.
I modified the code and got an error so I looked into it but couldn't figure it out.
 COPYING TO ROUTER dhd.ko => /jffs/dhd.ko
admin@192.168.50.1's password: 2024/07/04 15:31:51 socat[17267] E connect(6, AF=2 192.168.50.1:65535, 16): Connection refused
make: *** [Makefile:365: install-firmware] error 1

I have tried both wired and wireless with the same result.

Translated with DeepL.com (free version)

huiwenzhang1024, 2024-09-18:
> Sorry, I have been trying to find out how to fix the code. I modified the code and got an error so I looked into it but couldn't figure it out. COPYING TO ROUTER dhd.ko => /jffs/dhd.ko [...]

Hi, I have the same problem, have you solved that?