seemoo-lab/nexmon_csi #342: 2.4GHz CSI Extraction

sarielkk, 2024-05-15:
May I ask why the CSI rate collected in 2.4GHz is extremely slow? There may only be 2 data packets in more than ten seconds. When I use **ping** and **iperf**, the number of packets collected per unit time changes rapidly, and the sampling rate is extremely unstable. Is this normal?

jlinktu, 2024-05-15:
The 2.4 GHz band is way more crowded that the 5 GHz band. You might be on a channel with high interference and a high number of collisions, leading to a low number of correctly received frames and therefore to a low extraction rate. I recommend to you to choose a channel with low interference. You can also check with a normal monitor interface how the traffic looks like. CSI is extracted for every frame that is correctly received and matches your filter settings.

sarielkk, 2024-05-16:
> 2.4 GHz 频段比 5 GHz 频段更加拥挤。您可能处于具有高干扰和大量冲突的信道上，导致正确接收的帧数量较少，从而导致提取率较低。我建议您选择干扰较小的频道。您还可以使用普通监控界面检查流量情况。为正确接收并匹配您的过滤器设置的每个帧提取 CSI。

Thank you for your reply. I have successfully implemented CSI collection in monitoring mode at 5GHz, and the result is that 10 data packets were collected in 1 second.
I also configured frame injection (_nexmon_tx_task_) on the asus router, but I found that when I successfully started the frame injection mode and tx_delay was set to 500 milliseconds, the terminal showed "**done**", but from the data results collected by nexus6p, the collection is still 10 data packets were collected in 1 second.
Does this result mean that my frame injection is not set up successfully?

sarielkk, 2024-05-20:
> The 2.4 GHz band is way more crowded that the 5 GHz band. You might be on a channel with high interference and a high number of collisions, leading to a low number of correctly received frames and [...]

Thank you for your reply. I have sloved the problem.

closed 2024-05-20 by sarielkk

jlinktu, 2024-05-20:
Great to hear. As always, this thread would benefit from a description of your solution.