seemoo-lab/nexmon_csi #346: Operating system unable to install (Raspi 4B)

Asafe33, 2024-06-03:
Hello everyone! I've been trying to use a Raspberry Pi 4+ with Nexmon for a few months, but I keep encountering new errors. I've documented all the steps I've taken and will make them available at the link below for potential solutions.

I have tried the following OS versions: 2020-08-20, 2021-01-11, 2020-02-13, and 2022-01-28. The 2021-01-11 version was the most successful for me; however, I still couldn't capture the CSI data because nexutil failed, as you can see in the step-by-step guide in the link.

The 2022-01-28 version even started up normally, but it doesn't boot correctly and prompts for a username and password, which are not provided in its repository: https://github.com/nexmonster/nexmon_csi/tree/pi-5.10.92. If anyone knows how to work around these issues, I would be very grateful.

Errors: "start4.elf not compatible" in almost all versions.

Link: https://docs.google.com/document/d/1ZKlPFIoPvyCegAFUpnQTch8DGcvifu-wq-791A4Kw6E/edit?usp=sharing

Thank you for your help!

zeroby0, 2024-06-04:
There seems to be some error in step 4 with apt install in the document linked.

Try https://github.com/nexmonster/nexmon_csi_bin

Flash the 5.10.92 version raspbian and then run the command below it.

Asafe33, 2024-06-04:
Hello @zeroby0 , I followed your instructions, but the same screen still appears. Regardless of how long I leave the Raspberry Pi on, it remains stuck on the same screen. If I press any key, it prompts for a username and password, but the default password doesn't work. I flashed version 5.10.92 but haven't had any success so far.

[image]

zeroby0, 2024-06-04:
Click the return key, it should ask for login. Type `pi` and when it asks for password, type `raspberry`

Asafe33, 2024-06-04:
The instructions worked this time! The command prompt opened, asking me to configure the Wi-Fi, but the system didn't fully start. It said that "Wi-Fi is currently blocked by rfkill."

[image]

zeroby0, 2024-06-04:
From this point I think you can rely on Google for general issues :)

closed 2024-06-04 by jlinktu