seemoo-lab/nexmon_csi #347: The nexutil doesn't work

Asafe33, 2024-06-04:
Hey everybody. I've been working with Wi-Fi CSI for a few months, but I'm having trouble capturing CSI data. I followed the steps provided by @zeroby0 and installed from https://github.com/nexmonster/nexmon_csi_bin/.

However, at this point, I encountered a difficulty (see the picture). Nexutil is installed, but it's showing this error.

I'm using VirtualBox on my laptop to emulate Ubuntu, and wlan0 has changed to enp0s3 (TCPDUMP).

This error seems similar to one I encountered on 2021-01-11-raspios-buster-armhf-full (see picture 2). In both cases, I've installed Nexmon (disassembling ucode shows).

[image]
[image]
[image]
[image]

The pictures below are from Kernel 5.4.83-v8+. The backgrounds you see are from this kernel (2,3,4), but the other one is from Kernel 5.10.92 (1).

If anyone has a solution, I'd really appreciate it. 

@jlinktu @zeroby0

Asafe33, 2024-06-04:
I didn't find any solutions in the previous ones.

zeroby0, 2024-06-05:
Maybe you're substituting lowercase letter L for 1 in -l34. Copy paste instead of typing.

Asafe33, 2024-06-06:
Thank you, now it works!

closed 2024-06-06 by Asafe33