seemoo-lab/nexmon_csi #349: Is there a replacement for the ASUS AC86U? It's unavailable in Brazil

Asafe33, 2024-06-13:
Hey everyone. I recently noticed some discussions about using NEXUS and ASUS AC86U. Unfortunately, both are unavailable in Brazil. Are there any alternatives?

Would it be the latest version? The setup would include an RT laptop and a Raspberry Pi?

Thanks to the developers for creating this tool!

jlinktu, 2024-06-13:
A new version supporting recent devices is in the making. You still need some patience though.
Right now, the four devices listed on the README are the only devices we support with this repository.
If you don't need multiple antennas/streams, using a Raspberry Pi is currently the best option.

Asafe33, 2024-06-13:
Hi Jakob, thanks for the quick reply. I'll close that question here (:

closed 2024-06-13 by Asafe33

JackHack96, 2025-03-29:
I'm also interested in trying out this project, but the AC86U is difficult to purchase as it's out of stock in every shop in Norway...