seemoo-lab/nexmon_csi #350: Cannot capture csi packet  on AC86U

lionlanlan, 2024-06-17:
Greetings，
I have passed all steps on getting started and now try to capture packet from 149/80 channel（We use two other machine transport packet on this channel） but get nothing after running a long time.

makecsiparams -c 149/80 -C 1 -N 1 -m 8e:7c:36:a:21:8c,b0:a4:60:97:2c:28
admin@RT-AC86U-5360:/jffs#export PATH=$PATH:/jffs
admin@RT-AC86U-5360:/jffs#ifconfig eth6 up
admin@RT-AC86U-5360:/jffs#nexutil -Ieth6 -s20 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
admin@RT-AC86U-5360:/jffs#usr/sbin/wl -i eth6 monitor 1
admin@RT-AC86U-5360:/jffs# tcpdump -i eth6 dst port 5500
tcpdump: WARNING: eth6: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth6, link-type EN10MB (Ethernet), capture size 65535 bytes
^C
0 packets captured
6 packets received by filter
0 packets dropped by kernel

and if I skip the dst port, like 
admin@RT-AC86U-5360:/jffs# tcpdump -i eth6
 I can successfully capture a lot of packets, however, I cannot find where the captured packets are  and do not sure they are what I want.

jlinktu, 2024-06-18:
The following command you are using is wrong:
```
admin@RT-AC86U-5360:/jffs#nexutil -Ieth6 -s20 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
```
As described in the README the correct format would be:
```
nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
```

lionlanlan, 2024-06-18:
> The following command you are using is wrong:
>
> ```
> admin@RT-AC86U-5360:/jffs#nexutil -Ieth6 -s20 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
> ```
>
> As described in the README the [...]

yes I noticed it but if I set the s to 500：
admin@RT-AC86U-5360:/tmp/home/root# ifconfig eth6 up
admin@RT-AC86U-5360:/tmp/home/root# nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCO
fDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
__nex_driver_io: error ret=-1 errno=95

jlinktu, 2024-06-18:
Maybe you haven't loaded the `nexmon_csi` patched firmware.
You can check with `nexutil -Ieth6 -V | grep -ic nexmon`. If this produces `0`, you haven't loaded the `nexmon_csi` firmware. If that is the case, you can load it with `/sbin/rmmod dhd; /sbin/insmod /jffs/dhd.ko`. Afterwards, you need to configure the interface again.

lionlanlan, 2024-06-25:
> Maybe you haven't loaded the `nexmon_csi` patched firmware. You can check with `nexutil -Ieth6 -V | grep -ic nexmon`. If this produces `0`, you haven't loaded the `nexmon_csi` firmware. If that is [...]

hi，now I can run this command successfully 
 nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
but 
nexutil -Ieth6 -V | grep -ic nexmo
still produces 0.

lionlanlan, 2024-06-25:
> Maybe you haven't loaded the `nexmon_csi` patched firmware. You can check with `nexutil -Ieth6 -V | grep -ic nexmon`. If this produces `0`, you haven't loaded the `nexmon_csi` firmware. If that is [...]

hi，now I can run this command successfully
nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
but
nexutil -Ieth6 -V | grep -ic nexmon
still produces 0.

jlinktu, 2024-06-25:
What does `nexutil -Ieth6 -V` produce?

lionlanlan, 2024-06-26:
here is the nexutil -Ieth6 -V poduce：
admin@RT-AC86U-5360:/jffs# nexutil -Ieth6 -V
firmware 10.10.122.20 (r683106) FWID 01-9ac67028
vendorid 0x14e4
deviceid 0x43c5
radiorev 0x2103eb
chipnum 0xaa90
chiprev 0x4
chippackage 0x0
corerev 0x41
boardid 0x798
boardvendor 0x14e4
boardrev P102
driverrev 0xa0a7a14
ucoderev 0x46843fd
bus 0x0
phytype 0xb
phyrev 0x21
anarev 0x0
nvramrev 0x0

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 10.10.122.20 (r683106) FWID 01-9ac67028 |   0x14e4 |   0x43c5 |   0x2103eb |  0xaa90 |     0x4 |         0x0 |    0x41 |   0x798 |      0x14e4 |     P102 | 0xa0a7a14 | 0x46843fd | 0x0 |     0xb |   0x21 |    0x0 |      0x0

and another command get this：
admin@RT-AC86U-5360:/sbin# dmesg | grep 10.10.122.20
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43666 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.763 10.10.122.20 (nexmon.org/csi: fdb2-14)
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43664 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.774 10.10.122.20 (nexmon.org/csi: fdb2-14)

lionlanlan, 2024-06-26:
> What does `nexutil -Ieth6 -V` produce?

here is nexutil -Ieth6 -V produce：
admin@RT-AC86U-5360:/jffs# nexutil -Ieth6 -V
firmware 10.10.122.20 (r683106) FWID 01-9ac67028
vendorid 0x14e4
deviceid 0x43c5
radiorev 0x2103eb
chipnum 0xaa90
chiprev 0x4
chippackage 0x0
corerev 0x41
boardid 0x798
boardvendor 0x14e4
boardrev P102
driverrev 0xa0a7a14
ucoderev 0x46843fd
bus 0x0
phytype 0xb
phyrev 0x21
anarev 0x0
nvramrev 0x0

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 10.10.122.20 (r683106) FWID 01-9ac67028 |   0x14e4 |   0x43c5 |   0x2103eb |  0xaa90 |     0x4 |         0x0 |    0x41 |   0x798 |      0x14e4 |     P102 | 0xa0a7a14 | 0x46843fd | 0x0 |     0xb |   0x21 |    0x0 |      0x0

and another command get this：
admin@RT-AC86U-5360:/sbin# dmesg | grep 10.10.122.20
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43666 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.763 10.10.122.20 (nexmon.org/csi: fdb2-14)
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43664 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.774 10.10.122.20 (nexmon.org/csi: fdb2-14)

lionlanlan, 2024-06-26:
all my steps are：
admin@RT-AC86U-5360:/tmp/home/root# cd /sbin/
admin@RT-AC86U-5360:/sbin# rmmod dhd
admin@RT-AC86U-5360:/sbin# insmod /jffs/dhd.ko
admin@RT-AC86U-5360:/sbin# 
admin@RT-AC86U-5360:/sbin# dmesg | grep 10.10.122.20
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43666 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.763 10.10.122.20 (nexmon.org/csi: fdb2-14)
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43664 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.774 10.10.122.20 (nexmon.org/csi: fdb2-14)
admin@RT-AC86U-5360:/sbin# wl -i eth6 up
admin@RT-AC86U-5360:/sbin# wl -i eth6 radio on
admin@RT-AC86U-5360:/sbin# wl -i eth6 country US
admin@RT-AC86U-5360:/sbin# ifconfig eth6 up
admin@RT-AC86U-5360:/sbin# cd 
admin@RT-AC86U-5360:/tmp/home/root# cd /jffs/
admin@RT-AC86U-5360:/jffs# nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywp
GCXLCgAAAAAAAAAAAAAAAAAAA==
admin@RT-AC86U-5360:/jffs# nexutil -Ieth6 -V

jlinktu, 2024-06-26:
Ok, so apparently we do not overwrite this value, sorry for the confusion.
Anyways, the filtered output of `dmesg` confirms that you have loaded the correct firmware:
```
admin@RT-AC86U-5360:/sbin# dmesg | grep 10.10.122.20
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43666 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.763 10.10.122.20 (nexmon.org/csi: fdb2-14)
CONSOLE: RTE (PCIE-MSG_BUF) 10.10.122.20 (r683106) on BCM43664 r4 @ 40.0/200.0/800.0MHz
CONSOLE: 026738.774 10.10.122.20 (nexmon.org/csi: fdb2-14)
```
Also, there seems to be no error when you execute
```
admin@RT-AC86U-5360:/jffs# nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA==
```
Which means, you have configured the CSI extractor.
To extract CSI you are missing one more step, activation of monitor mode: `/usr/sbin/wl -i eth6 monitor 1` (Please check the README again!)

maple-42, 2024-08-01:
Hi,
When I tried to capture CSI, I encountered the same error and was able to solve the error with the help of the solution here.
However, after that
tcpdump -i wlan0 dst port 5500
When I run this, I get
Permission denied
error.
Please let me know if you have a solution.

jlinktu, 2024-08-01:
You have to execute this on the router.

maple-42, 2024-08-01:
Thanks for the reply.
I run this on my router and get this error.

jlinktu, 2024-08-01:
On the router there is no interface called `wlan0`. Please head back to the README and read it carefully.

jlinktu, 2024-08-01:
`tcpdump` is not installed by default on the router, have you cross-compiled it correctly?
See this post for mor info: https://github.com/seemoo-lab/nexmon_csi/issues/34#issuecomment-577374225

maple-42, 2024-08-01:
I thought I had cross-compiled correctly, but maybe I did not.
I will try again later and get back to you.
Thank you very much.

maple-42, 2024-08-02:
Hi,
I just tried it and I get the same error.

admin@RT-AC86U:/jffs# ls
dhd.ko iperf3 nexutil nmp_cl_json.js nmp_client_list syslog.log tcpdump
admin@RT-AC86U:/jffs# wl -i eth6 up
admin@RT-AC86U:/jffs# wl -i eth6 radio on
admin@RT-AC86U:/jffs# wl -i eth6 chanspec 161/80
Chanspec set to 0xe39b
admin@RT-AC86U:/jffs# wl -i eth6 monitor 1
admin@RT-AC86U:/jffs# ifconfig eth6 up
admin@RT-AC86U:/jffs# tcpdump -i eth6 -v dst port 5500
-sh: tcpdump: not found
admin@RT-AC86U:/jffs# . /tcpdump -i eth6 -v dst port 5500
-sh: . /tcpdump: Permission denied

This is what I have run.
Thank you very much.

jlinktu, 2024-08-02:
Have you even set execution rights on `tcpdump`? If not: 
`chmod u+x /jffs/tcpdump`

jlinktu, 2024-08-02:
No, on the router.

maple-42, 2024-08-02:
Okay, I will try it.
Thanks for the reply.

maple-42, 2024-08-03:
I ran the above command and successfully ran tcpdump. Thank you very much.
However, it seems that no packets are being captured.

admin@RT-AC86U:/jffs# . /nexutil -Ieth6 -s500 -b -l34 -vm+MBEQGIAQAE1MRDyLgAAAAAAAAAAAAAAAAAAAAA==
admin@RT-AC86U:/jffs# wl -i eth6 up
admin@RT-AC86U:/jffs# wl -i eth6 radio on
admin@RT-AC86U:/jffs# wl -i eth6 chanspec 161/80
Chanspec set to 0xe39b
admin@RT-AC86U:/jffs# wl -i eth6 monitor 1
admin@RT-AC86U:/jffs# ifconfig eth6 up
admin@RT-AC86U:/jffs# . /tcpdump -i eth6 -v dst port 5500 -w ~/pcap/1.pcap -c 1000
tcpdump: WARNING: eth6: no IPv4 address assigned
tcpdump: /root/pcap/1.pcap: No such file or directory
admin@RT-AC86U:/jffs# . /tcpdump -i eth6 -v dst port 5500 -w /tmp/1.pcap -c 1000
tcpdump: WARNING: eth6: no IPv4 address assigned
tcpdump: listening on eth6, link-type EN10MB (Ethernet), capture size 65535 bytes
^C0 packets captured
0 packets received by filter
0 packets dropped by kernel

I don't know what is causing this. Thank you in advance.

maple-42, 2024-08-03:
eth6 Link encap:Ethernet HWaddr 04:D4:C4:43:C8:BC  
          UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
          RX packets:5545 errors:0 dropped:6 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1604105 (1.5 MiB) TX bytes:0 (0.0 B)

This is the result of running ifconfig.

eth6 Link encap:Ethernet HWaddr 04:D4:C4:43:C8:BC  
          inet addr:192.168.50.100 Bcast:192.168.50.255 Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
          RX packets:5549 errors:0 dropped:7 overruns:0 frame:0
          TX packets:0 errors:0 dropped:23 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1605243 (1.5 MiB) TX bytes:0 (0.0 B)

The result was the same when I ran it with the ip address assigned in this way.

maple-42, 2024-08-03:
admin@RT-AC86U:/jffs# . /tcpdump -i eth6 -v
Thus, if we did not specify the port, we were able to capture it.
Does this not allow me to get CSI?

abeyudai-tohoku, 2025-01-10:
> hi，now I can run this command successfully nexutil -Ieth6 -s500 -b -l34 -vm+ABEQAAAgCOfDYKIYywpGCXLCgAAAAAAAAAAAAAAAAAAA== but nexutil -Ieth6 -V | grep -ic nexmon still produces 0

Greetings，
Could you tell me how to run that command successfully??
I have same problem.